intel/media-driver #674: media server studio 2016 support for haswell

ulyuing, 2019-07-11:
hi , I recently try to get my E5-1200 V3 support for media sdk，but only the old drvier can do this job(media server studio 2016).  however, I failed to find out download link(only release note can be find). is there anyway to get this driver? thank you!

XinfengZhang, 2019-09-02:
sorry for late response, but actually, I am not clearly understand your question. you want a older version msdk ?

onabiull, 2019-09-27:
Unfortunately we have no support for E3-1200 V3 (I guess this is typo, as E5 Xeons have no integrated graphics at all) in open source space. Relatively old releases (say 1.1a, 1.2a) of Media SDK still have gen7.5 support, but was not validated and I can't recommend them. Theoretically combination of older i965 media driver and older Media SDK may work, but that combination was not tested at all. I suspect that features like basic decode may be functional, but highly suspect that more complex scenarios would fail.
If you are looking for exactly MSS 2016 binaries, I'd suggest to contact support channel.

dvrogozh, 2019-09-27:
If trying to make something alive for haswell, I would try latest mediasdk release with the latest i965 driver. As Oleg mentioned, decoding will likely work or require pretty small effort to fix. Encoding/VP might be a challenge.

closed 2019-09-29 by ulyuing

augix, 2022-12-28:
> sorry for late response, but actually, I am not clearly understand your question. you want a older version msdk ?

where can I find a old version of media server studio? I have an integrated graphics card: Intel HD graphics 4400. I would like to make opencl work for this card.