intel/media-driver #677: [ICL] GPU hangs on hevc y410 (ayuv) encoding in VDEnc mode

AntonGrishin, 2019-07-11:
Environment:

- ICL linux
- Q2 release versions of [linux media stack ](https://github.com/Intel-Media-SDK/MediaSDK/releases/tag/intel-mediasdk-19.2.0).

cmd: `./sample_encode h265 -y410 -qsv-ff -i stream176x144_y410.yuv -w 176 -h 144 -o ./stream176x144.hevc -n 5`

Expected result:
>Processing started
>Frame number: 5
>Processing finished

Actual results:
> Processing started
> Frame number: 1
> [ERROR], sts=MFX_ERR_GPU_HANG(-21), SynchronizeFirstTask, SyncOperation failed at /localdisk/bb/worker/build-api-next/build_dir/repos/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:157
> 
> [ERROR], sts=MFX_ERR_GPU_HANG(-21), Run, m_TaskPool.SynchronizeFirstTask failed at /localdisk/bb/worker/build-api-next/build_dir/repos/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:2353
> 
> [ERROR], sts=MFX_ERR_GPU_HANG(-21), main, pPipeline->Run failed at /localdisk/bb/worker/build-api-next/build_dir/repos/MediaSDK/samples/sample_encode/src/sample_encode.cpp:1587

If build driver from master branch, then encode passes. 
Fixed commit from master: [a674421]( https://github.com/intel/media-driver/commit/a674421e8aa5fb0b17a49446c54b0080ee07999e)

AntonGrishin, 2019-07-11:
@Sherry-Lin @onabiull @vvkalinin24

Sherry-Lin, 2019-08-01:
resolved by https://github.com/intel/media-driver/commit/6cc99540, so close it.

closed 2019-08-01 by Sherry-Lin