intel/media-driver #678: vaQueryImageFormats missing packed RGB24 format in postproc

HeJunyan, 2019-07-11:
Gstreamer

gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=RGB,width=800,height=600 ! vaapipostproc ! video/x-raw,format=NV12 ! filesink location=ppp.yuv

can not work, because the FOURCC of VA_FOURCC_R8G8B8 can not be found in current driver.

While, I checked, at least the version:

commit d17f3fcf7e5238ae82b50c54c9591cde919ae3fe 
Author: Huang_LiH <li.h.huang@intel.com>
Date:   Wed May 22 08:34:58 2019 +0800

    [CM] surface dump enhance

    Enhance surface dump

    Change-Id: I18fa07fd2714efe30518cf75e287245382ca49aa

still have this support

Jister, 2019-07-12:
Since driver cannot output R8G8B8, currently, we cannot separate the image capability for input and output, and we remove it in image capability immediately, but you still can use it as input.

FurongZhang, 2019-07-12:
Jister, please help to follow up this issue, thanks a lot in advance!

XinfengZhang, 2020-03-19:
it is a limitation of caps reporting, could we close it or move it to libva @HeJunyan