intel/media-driver #679: vaQueryImageFormats return wrong format of AYUV, and failed to create its image

HeJunyan, 2019-07-11:
vaQueryImageFormats of AYUV format return result
byte_order 1, bits_per_pixel 32, depth 24, red_mask ff0000, green_mask ff00, blue_mask ff, alpha_mask ff000000,

which is totally wrong, it's a YUV format and should have no RGB mask
Also failed to create this format's image, please check.

At least, the commit:

commit d17f3fcf7e5238ae82b50c54c9591cde919ae3fe (HEAD -> master)
Author: Huang_LiH <li.h.huang@intel.com>
Date:   Wed May 22 08:34:58 2019 +0800

    [CM] surface dump enhance

    Enhance surface dump

    Change-Id: I18fa07fd2714efe30518cf75e287245382ca49aa

Still work well. 
Please check it

FurongZhang, 2019-07-12:
I will fix it.

FurongZhang, 2019-07-12:
https://github.com/intel/media-driver/pull/681/commits/bde9f6eec5d40dd1e9841ee761302fba7c1202df

FurongZhang, 2019-07-12:
Please verify the above PR. Sorry for the issue.

XinfengZhang, 2020-05-27:
as comments from Furong, it should already been fixed by https://github.com/intel/media-driver/commit/bde9f6eec5d40dd1e9841ee761302fba7c1202df

closed 2020-05-27 by XinfengZhang