intel/media-driver #686: [ICL VPP]  brightness affect UV on ICL

zhuqingliang, 2019-07-16:
description:
When do vpp brightness, the UV will change on ICL , but KBL will not change UV

reproduce step:
set brightness=0
1.  ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug -f rawvideo -pix_fmt yuv420p -s:v 1280x720 -i /root/media/testing/assets/yuv/720p5994_parkrun_ter_1280x720_I420.yuv -vf 'format=nv12,hwupload,procamp_vaapi=b=-100.0,hwdownload,format=nv12' -pix_fmt yuv420p -vframes 10 -y test.yuv
2.   compare psnr between test.yuv and 720p5994_parkrun_ter_1280x720_I420.yuv, UV will not change on KBL, but ICL changed.

Alex1Zhang, 2020-02-25:
This is an expected behavior. Because ProCamp support 4:4:4 and for NV12(4:2:0) driver need setup ChromaSampling on ICL, so the UV value will change. On KBL the ChromaSampling is operated in HW with duplicate value, so the UV value can not change. Thanks.

XinfengZhang, 2020-03-19:
@zhuqingliang does the reply  make sense to you? if yes, could you close the issue?

XinfengZhang, 2020-04-16:
@Alex1Zhang , I am confused. it is HW restriction ? we should follow the spec, brightness change should only impact Y Luma part, right?

FurongZhang, 2020-04-23:
As for the current API setting, the output is expected due to HW implementation.

FurongZhang, 2020-04-24:
The below is the calculation of ProcAmp.
Yout' = ((Yin-256) x C) + B + 256,
Uout' = (((Uin-2048) x Cos(H) + (Vin-2048) x Sin(H)) x C x S) + 2048
Vout' = (((Vin-2048) x Cos(H) - (Uin-2048) x Sin(H)) x C x S) + 2048

If Setting B as 0,  from the calculation formula, UV should not be changed. 

But in VEBOX, we use 444 to processing data. So Chroma Upsampling is needed from 420 to 444. In media driver, the default chroma up sampling sitting is Horizontal Left, Vertical Center (for this setting, VEBOX will do a bilinear interpolation). This default value is from video spec(please check AVC spec or HEVC spec). If you don’t want to sit as the default, VAAPI user needs to set Chroma Sitting explicitly as Horizontal left and Vertical Top.

FurongZhang, 2020-04-24:
@xuguangxin has a comment. It is a valid comment. I need to thank about this.
Hi Furong, 
Upsampling is driver/hw internal implementation. A user can’t know driver/hw will do 420 -> 444 conversion. 
Why do we ask a user to set the chrome sitting for brightness?

All user’s asking is brightness adjustment not impact uv.
Could we hide the chrome sitting thing in the driver?

thanks

uartie, 2020-04-24:
> The below is the calculation of ProcAmp.
> Yout' = ((Yin-256) x C) + B + 256,
> Uout' = (((Uin-2048) x Cos(H) + (Vin-2048) x Sin(H)) x C x S) + 2048
> Vout' = (((Vin-2048) x Cos(H) - (Uin-2048) x Sin(H)) x C x S) + 2048
> 
> If Setting B as 0, from the calculation formula, UV should not be changed.

Huh?  I'm confused by this last statement?  `B` is not even a variable in the U and V formula/equations.  So, regardless of any value for `B`, it should not have any effect on the U and V formula.

FurongZhang, 2020-05-11:
For our perspective, this is not an issue. It is expected.

FurongZhang, 2020-05-11:
Since it is expected, we are put that into low priority.

FurongZhang, 2020-05-26:
From my perspective, I think the output is trending to be expected and reject to make changes in driver due to the below:
1.	Low ROI;
2.	Current driver behavior is expected and consistent with Video Spec; 
3.	It is due to HW up sampling UV to 444;

I am requesting the VAAPI user to set to Top, Left for chroma sitting. Please see the definition in va_vpp.h.

 /** @name Chroma Siting flag */
/**@{*/
/** vertical chroma sitting take bit 0-1, horizontal chroma sitting take bit 2-3
 * vertical chromma siting | horizontal chroma sitting to be chroma sitting */
#define VA_CHROMA_SITING_UNKNOWN              0x00
/** \brief Chroma samples are co-sited vertically on the top with the luma samples. */
#define VA_CHROMA_SITING_VERTICAL_TOP         0x01
/** \brief Chroma samples are not co-sited vertically with the luma samples. */
#define VA_CHROMA_SITING_VERTICAL_CENTER      0x02
/** \brief Chroma samples are co-sited vertically on the bottom with the luma samples. */
#define VA_CHROMA_SITING_VERTICAL_BOTTOM      0x03
/** \brief Chroma samples are co-sited horizontally on the left with the luma samples. */
#define VA_CHROMA_SITING_HORIZONTAL_LEFT      0x04
/** \brief Chroma samples are not co-sited horizontally with the luma samples. */
#define VA_CHROMA_SITING_HORIZONTAL_CENTER    0x08
/**@}*/

FurongZhang, 2020-05-26:
@Alex1Zhang , please double confirm this issue.

Alex1Zhang, 2020-06-03:
@xhaihao @xuguangxin please try #956 to test the ProCamp issue which only adjust Brightness. Thanks.

xuguangxin, 2020-06-03:
Thanks alex,
Hi @Zhang, YuankunX<mailto:YuankunX.Zhang@intel.com>, could you help test this patch on ICL.
see brightness uv change issue is fixed or not?
You can start at sanity test firstly, then full round test for vpp.

thanks

xuguangxin, 2020-06-04:
Hi guangxin,
因为yinhang在用我的电脑去配置直播的那个demo，所以我没有及时看到你的信息，我马上开始测试。
Thanks,
--yuankun

XinfengZhang, 2020-07-15:
fixed by #956

closed 2020-07-15 by XinfengZhang