intel/media-driver #690: [ffmpeg][vaapi][ARGB/BGRA] mismatch of ARGB/BRGA

fulinjie, 2019-07-23:
cmdline:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i Nature.h264 -vf scale_vaapi=format=rgb32,hwdownload,format=rgb32 -f rawvideo /dev/null

https://github.com/FFmpeg/FFmpeg/blob/3883c9d1479d3e653bb66c9960fdd2ebd6ba9594/libavutil/hwcontext_vaapi.c#L607
if (expected_format->fourcc == test_image.format.fourcc) {

expected_format->fourcc is different from  test_image.format.fourcc.

There is mismatch between ffmpeg and media driver for ARGB/BGRA, thus vaDeriveImage can't be used, and it causes the performance drop.

wangyan-intel, 2019-07-23:
@fulinjie Could you share your video clip for debugging? Thanks.

fulinjie, 2019-07-23:
@wangyan-intel Random clips should be ok, since the cmdline simply decoded a clips and do color space convert.

wangyan-intel, 2019-07-24:
I tried it by one MP4 h264 clip but cannot find this issue. the commit of iHD driver is c2aae3e8. Could you please double check it? Thanks.

fulinjie, 2019-07-24:
Hi Yan, update to latest version of iHD driver, and this issue still exists.
This happens when vaapi_frames_init is called for VPP (csc, called the second time.)

vaCreateImage is used instead of vaDeriveImage, thus makes the performance drop.

wangyan42164, 2019-07-24:
@fulinjie Could you please try #692? It works fine on my side. Thanks.

fulinjie, 2019-08-15:
Hi @wangyan42164, how is this going?

wangyan-intel, 2020-03-11:
@XinfengZhang As I know, @Jister is working on this?

MicroYY, 2022-01-18:
We're trying @wangyan42164 's fix.  As it's an old PR and contains confilct, @wangyan42164 could you pls rebase the patch?

MicroYY, 2022-03-15:
@LhGu What's the status of your PR?

LhGu, 2022-03-15:
> @LhGu What's the status of your PR?

It is under review.

LhGu, 2022-03-18:
There is a limitation in the driver stack. It only supports the hex order is B-G-R-A, and our driver treats the hex order as VA_FOURCC_ARGB format.  I already have two PRs（#1334， #1335 ）to work around it.  
However, **any work around code may be confused and affect our internal test cases**.  Just for this issue, **you can use this FFmpeg command to get the same output**:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i Nature.h264 -vf scale_vaapi=format=**argb**,hwdownload,format=**argb** -f rawvideo /dev/null

This command output is the same as original command (B-G-R-A) , and its VA_FOURCC will be consistent with FFmpeg.

This case just requests the read access,  and prefers not to be given direct-mapped memory. The reason has been explained in the below link: [https://github.com/FFmpeg/FFmpeg/blob/3883c9d1479d3e653bb66c9960fdd2ebd6ba9594/libavutil/hwcontext_vaapi.c#L780]
If you want to test the HW map, you need to change this judgement code in FFmpeg, besides the consistent VA_FOURCC. 

I think that we can close this issue with no driver code changed.

LhGu, 2022-05-16:
This issue needs feedback, but no feedback from submitter yet. I will close this issue if no more information provided from submitter in 1 month. Please feel free to re-open this issue if it still exist.

LhGu, 2022-05-31:
Close it now, and please feel free to re-open this issue if you want to discuss more.

closed 2022-05-31 by LhGu