intel/media-driver #695: [ffmpeg-vaapi] VC1 decode failed

intel-media-ci, 2019-07-25:
The commit ID bb920ee213275d234fbfb6d6eb868a10bf0a5274 cause VC1 decode failed.
It looks like a regression.
https://github.com/intel/media-driver/commit/bb920ee213275d234fbfb6d6eb868a10bf0a5274

The same command line works for the previous commit ID
https://github.com/intel/media-driver/commit/a798a38fff76ae738188813a4cb416c0204fb0da

The command line like following - 
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -filter_hw_device hw -v verbose -i /opt/media/src/assets/otc-media/vc1/SA10090.vc1 -pix_fmt yuv420p -f rawvideo -vsync passthrough -vframes 5 -y /opt/media/test-results/vaapi-fits/ffmpeg-vaapi-iHD/f9327658-ade4-11e9-a61f-0242ac110002_0/_0.test.ffmpeg-vaapi.decode.vc1/default/SA10090_720x480_I420.yuv

[sysout4108848447859486443.txt](https://github.com/intel/media-driver/files/3430235/sysout4108848447859486443.txt)

https://media-ci.ostc.intel.com/build/306843   VC1 decode Passed
https://media-ci.ostc.intel.com/build/306898   VC1 decode failed

Please let me know if you need more information.

wangyan-intel, 2019-07-25:
@maksenox Could you help check it? Thanks.

Jexu, 2021-04-14:
the issue looks gone on latest media-driver, pls help confirm that, then we can close this issue.

Jexu, 2021-05-11:
hi @XinfengZhang , can you help close this issue since it has no update for a long time. 
above commit causes failure to free surface, but i don't see the issue in latest driver. i suspect the issue has been fixed somewhere.

closed 2021-05-14 by XinfengZhang