intel/media-driver #700: [ffmpeg_vaapi][hevc_lp_encode] encode file has garbage on ICL

zhuqingliang, 2019-08-05:
description:
  the color have some error on hevc lowpower encode cbr/vbr mode

reproduce step:
1.ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v 352x288 -r:v 30 -i yuv/MOBILE_352x288_30_orig_01_352x288_I420.yuv -vf 'format=nv12,hwupload' -c:v hevc_vaapi -profile:v main -rc_mode CBR -g 30 -slices 1 -b:v 100k -maxrate 100k -low_power 1 -vframes 300 -y test.h265

2. observe
the color have some garbage, screen shot below 
[image]
[image]

wangyan-intel, 2019-08-12:
@zhuqingliang Could you share your video clip for debugging? Thanks.

zhuqingliang, 2019-08-14:
@wangyan-intel  https://drive.google.com/file/d/1mnTYUS_0MMGT0az4MbihvIL_BmyGas-T/view?usp=sharing

wangyan-intel, 2020-05-13:
This is encoding issue.

Xiaogangli-intel, 2020-07-01:
@zhuqingliang This issue should be fixed in the latest driver? Could you have a try?
And please use this FFmpeg which has some fixes for VAAPI.
PS: Please run _git am patchset/*.patch_ before build.

Xiaogangli-intel, 2020-07-09:
This issue has been fixed in the latest driver, so close it.

closed 2020-07-09 by Xiaogangli-intel