intel/media-driver #701: creating surfaces for I420 with VA_SURFACE_ATTRIB_MEM_TYPE_USER_PTR

shakeeb80, 2019-08-10:
I am trying to use vaCreateSurfaces to create an I420 surface with VA_SURFACE_ATTRIB_MEM_TYPE_USER_PTR memory type. However it looks like any multi plane surface is not supported.

VASurfaceAttribExternalBuffers extern_buf;
  memset(&extern_buf, 0, sizeof(VASurfaceAttribExternalBuffers));
  extern_buf.buffers = new unsigned long[1]; 
  extern_buf.buffers[0] = reinterpret_cast<unsigned long>(
      i420_data;
  extern_buf.num_buffers = 1;
  extern_buf.width =width;
  extern_buf.height = height;
  extern_buf.num_planes =3;
  extern_buf.data_size = video_info->size;
  for (int i = 0; i < 4; i++) {
    extern_buf.pitches[i] = video_info->stride[i];
    extern_buf.offsets[i] = video_info->offset[i];
  }
  extern_buf.pixel_format = pixel_format;

dvrogozh, 2019-08-14:
You asked this question in libva. Please, provide answer on questions from this comment: https://github.com/intel/libva/issues/326#issuecomment-520578078. Posting it here as well:

How original i420 memory was allocated? Mind that userptr path has 2 basic requirements:

Pointer should be page aligned
Data size should be page aligned
There are other requirements implied by the media-driver, but the above 2 are basic ones. As for the media-driver requirements, the following would likely be supreme requirements:

Width should be 16-aligned
Height should be 4-aligned
Pitch should be 64-aligned
I afraid, however, there could be some issues in a media-driver on the userptr path since this is rarely used path. Please, share anything you will come across with us.

XinfengZhang, 2020-07-02:
no response, the USER_PTR should works if the allocation of system memory is aligned with the requirements

closed 2020-07-02 by XinfengZhang