intel/media-driver #703: Support for Bay Trail?

tmm1, 2019-08-11:
On the J1900 CPU it seems intel-vaapi-driver works but intel-media-driver does not support it.

What is the process for adding support? Is this something that I could port from the vaapi-driver, or something that only Intel would be able to accomplish?

eero-t, 2019-08-13:
Is there a particular reason why you would want to use the new driver on the old HW instead of using the driver that was written for that HW?

(Even if somebody could port new driver to support also the old HW it's possible, probably even likely, that it would never support all the HW features that the old driver does.)

tmm1, 2019-08-13:
Just as a matter of convenience for programs/scripts that expect to run on multiple systems. It would be easier if the same driver and flags worked without having to probe the system and CPU first.

But I understand this is not feasible and that the legacy driver is needed for legacy chipsets.

closed 2019-08-13 by tmm1