intel/media-driver #704: Encoding failure on Kaby Lake

tmm1, 2019-08-11:
I'm getting failures on i5-7260U:

```
[MOS]:  CRITICAL - SubmitCommandBuffer:903: Command buffer submission failed!
[CODEC]:CRITICAL - SubmitCommandBuffer:4070: MOS returned error, eStatus = 0x20
[CODEC]:CRITICAL - ExecuteSliceLevel:6993: MOS returned error, eStatus = 0x20
[CODEC]:CRITICAL - ExecuteEnc:4414: Slice level encoding failed.
[CODEC]:CRITICAL - Execute:583: MOS returned error, eStatus = 0x20
[LIBVA]:CRITICAL - EncodeInCodecHal:1146: DDI:Failed in Codechal!
[LIBVA]:CRITICAL - EndPicture:81: DDI:DdiEncode_EncodeInCodecHal return failure.
[h264_vaapi @ 0x3a66cc0] Failed to end picture encode issue: 24 (internal encoding error).
[h264_vaapi @ 0x3a66cc0] Encode failed: -5.
```

What could cause this issue?

wangyan-intel, 2019-08-12:
@tmm1 Could you share your command and input file for debugging? Thanks.

tmm1, 2019-08-12:
>ffmpeg -t 3 -init_hw_device vaapi=intel:/dev/dri/renderD128 -hide_banner -loglevel verbose -i stream.mpg -c copy -filter_hw_device intel -vf "format=nv12,hwupload" -b:v 6000k -maxrate 8000k -c:v h264_vaapi -an -f h264 -y /dev/null

tmm1, 2019-08-12:
stream.mpg: http://0x0.st/zVAr.mpg

wangyan-intel, 2019-08-13:
@tmm1 Thanks for sharing. But I cannot download your video clip so far. Could you share it by the other method? Thanks.

tmm1, 2019-08-13:
How about this one: https://www.dropbox.com/s/k2xc5psgp3l6soh/stream.mpg?dl=1

If that does not work I can upload to S3.

wangyan-intel, 2019-08-13:
@tmm1 Sorry. I cannot access dropbox. Google Doc may be OK. Thanks.

tmm1, 2019-08-13:
https://drive.google.com/file/d/1faAXczONDfI65oT2akQF_Ed5Xt5YQbte/view?usp=drivesdk

wangyan-intel, 2019-08-13:
@tmm1 Get it. Thanks.

wangyan-intel, 2019-08-13:
@tmm1 I tried on my local platform  (Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz) with media driver (4ee2a4d8c) but I cannot find your mentioned error. I also checked the output h264 video clip. It is OK.
The following is the my ffmpeg log. Could you please try it with the latest media driver? Thanks.

[AVHWDeviceContext @ 0x43f66c0] libva: VA-API version 1.6.0
[AVHWDeviceContext @ 0x43f66c0] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x43f66c0] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x43f66c0] libva: Trying to open /usr/lib64/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x43f66c0] libva: Found init function __vaDriverInit_1_6
[AVHWDeviceContext @ 0x43f66c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x43f66c0] Initialised VAAPI connection: version 1.6
[AVHWDeviceContext @ 0x43f66c0] VAAPI driver: Intel iHD driver - 1.0.0.
[AVHWDeviceContext @ 0x43f66c0] Driver not found in known nonstandard list, using standard behaviour.
[mpeg2video @ 0x443c700] Invalid frame dimensions 0x0.
    Last message repeated 24 times
[mpegts @ 0x44380c0] PES packet size mismatch
    Last message repeated 1 times
Input #0, mpegts, from 'stream.mpg':
  Duration: 00:00:30.39, start: 27664.978322, bitrate: 10381 kb/s
  Program 3100 
    Stream #0:0[0x3e9]: Video: mpeg2video (Main), 1 reference frame ([2][0][0][0] / 0x0002), yuv420p(tv, progressive, left), 1280x720 [SAR 1:1 DAR 16:9], Closed Captions, 59.94 fps, 59.94 tbr, 90k tbn, 119.88 tbc
    Stream #0:1[0x3ea](eng): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, 5.1(side), fltp, 448 kb/s
    Stream #0:2[0x3eb](spa): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, stereo, fltp, 192 kb/s (visual impaired) (descriptions)
Stream mapping:
  Stream #0:0 -> #0:0 (mpeg2video (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[mpegts @ 0x44380c0] Correcting start time by 325534
[graph 0 input from stream 0:0 @ 0x4973a80] w:1280 h:720 pixfmt:yuv420p tb:1/90000 fr:60000/1001 sar:1/1 sws_param:flags=2
[auto_scaler_0 @ 0x45d9a40] w:iw h:ih flags:'bicubic' interl:0
[trim_in_0_0 @ 0x449b080] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'trim_in_0_0'
[auto_scaler_0 @ 0x45d9a40] w:1280 h:720 fmt:yuv420p sar:1/1 -> w:1280 h:720 fmt:nv12 sar:1/1 flags:0x4
[h264_vaapi @ 0x445dd00] Input surface format is nv12.
[h264_vaapi @ 0x445dd00] Using VAAPI profile VAProfileH264High (7).
[h264_vaapi @ 0x445dd00] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[h264_vaapi @ 0x445dd00] Using VAAPI render target format YUV420 (0x1).
[h264_vaapi @ 0x445dd00] RC mode: AVBR.
[h264_vaapi @ 0x445dd00] RC target: 6000000 bps converging in 120 frames with 100% accuracy.
[h264_vaapi @ 0x445dd00] RC framerate: 60000/1001 (59.94 fps).
[h264_vaapi @ 0x445dd00] Using intra, P- and B-frames (supported references: 8 / 2).
[h264_vaapi @ 0x445dd00] All wanted packed headers available (wanted 0xd, found 0x1f).
[h264_vaapi @ 0x445dd00] Using level 3.2.
Output #0, h264, to 'test.h264':
  Metadata:
    encoder         : Lavf58.28.102
    Stream #0:0: Video: h264 (h264_vaapi) (High), 1 reference frame, vaapi_vld(left), 1280x720 [SAR 1:1 DAR 16:9], q=-1--1, 6000 kb/s, 59.94 fps, 59.94 tbn, 59.94 tbc
    Metadata:
      encoder         : Lavc58.53.101 h264_vaapi
*** 28 dup!
No more output streams to write to, finishing.e=00:00:02.85 bitrate=5145.8kbits/s dup=28 drop=0 speed=1.89x    
frame=  208 fps=110 q=-0.0 Lsize=    2369kB time=00:00:03.45 bitrate=5620.6kbits/s dup=28 drop=0 speed=1.83x    
video:2369kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (stream.mpg):
  Input stream #0:0 (video): 209 packets read (4020884 bytes); 182 frames decoded; 
  Input stream #0:1 (audio): 0 packets read (0 bytes); 
  Input stream #0:2 (audio): 0 packets read (0 bytes); 
  Total: 209 packets (4020884 bytes) demuxed
Output file #0 (test.h264):
  Output stream #0:0 (video): 208 frames encoded; 208 packets muxed (2426291 bytes); 
  Total: 208 packets (2426291 bytes) muxed
[AVIOContext @ 0x4492b80] Statistics: 0 seeks, 10 writeouts
[AVIOContext @ 0x4411e40] Statistics: 9949328 bytes read, 2 seeks

tmm1, 2019-08-13:
My error was with version 19.2

I will try again with master

tmm1, 2019-08-13:
I rebuilt with f0874503a3b02c7af3ea92865e3309800fac4d17 (also had to upgrade to g++ 6.4.0), but I'm still having the same issue:

```
tmm1@nuc:~$ uname -a
Linux nuc 4.4.0-127-generic #153-Ubuntu SMP Sat May 19 10:58:46 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
tmm1@nuc:~$ ./ffmpeg-4.2-ihd-master -t 3 -init_hw_device vaapi=intel:/dev/dri/renderD128 -hide_banner -loglevel verbose -i stream.mpg -c copy -filter_hw_device intel -vf "format=nv12,hwupload" -b:v 6000k -maxrate 8000k -c:v h264_vaapi -level 42 -an -f h264 -y out.h264
[AVHWDeviceContext @ 0x48e8380] libva: VA-API version 1.5.0
[AVHWDeviceContext @ 0x48e8380] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x48e8380] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x48e8380] Initialised VAAPI connection: version 1.5
[AVHWDeviceContext @ 0x48e8380] VAAPI driver: Intel iHD driver - 1.0.0.
[AVHWDeviceContext @ 0x48e8380] Driver not found in known nonstandard list, using standard behaviour.
Input #0, mpegts, from 'stream.mpg':
  Duration: 00:00:30.39, start: 27664.978322, bitrate: 10381 kb/s
  Program 3100
    Stream #0:0[0x3e9]: Video: mpeg2video (Main), 1 reference frame ([2][0][0][0] / 0x0002), yuv420p(tv, progressive, left), 1280x720 [SAR 1:1 DAR 16:9], Closed Captions, 59.94 fps, 59.94 tbr, 90k tbn, 119.88 tbc
    Stream #0:1[0x3ea](eng): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, 5.1(side), fltp, 448 kb/s
    Stream #0:2[0x3eb](spa): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, stereo, fltp, 192 kb/s (visual impaired) (descriptions)
Stream mapping:
  Stream #0:0 -> #0:0 (mpeg2video (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[mpegts @ 0x491f940] Correcting start time by 325534
[graph 0 input from stream 0:0 @ 0x4ae4380] w:1280 h:720 pixfmt:yuv420p tb:1/90000 fr:60000/1001 sar:1/1 sws_param:flags=2
[auto_scaler_0 @ 0x4ae3900] w:iw h:ih flags:'bicubic' interl:0
[trim_in_0_0 @ 0x4ae4c40] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'trim_in_0_0'
[auto_scaler_0 @ 0x4ae3900] w:1280 h:720 fmt:yuv420p sar:1/1 -> w:1280 h:720 fmt:nv12 sar:1/1 flags:0x4
[h264_vaapi @ 0x4945cc0] Input surface format is nv12.
[h264_vaapi @ 0x4945cc0] Using VAAPI profile VAProfileH264High (7).
[h264_vaapi @ 0x4945cc0] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[h264_vaapi @ 0x4945cc0] Using VAAPI render target format YUV420 (0x1).
[h264_vaapi @ 0x4945cc0] RC mode: AVBR.
[h264_vaapi @ 0x4945cc0] RC target: 6000000 bps converging in 120 frames with 100% accuracy.
[h264_vaapi @ 0x4945cc0] RC framerate: 60000/1001 (59.94 fps).
[h264_vaapi @ 0x4945cc0] Using intra, P- and B-frames (supported references: 8 / 2).
[h264_vaapi @ 0x4945cc0] All wanted packed headers available (wanted 0xd, found 0x1f).
Output #0, h264, to 'out.h264':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: h264 (h264_vaapi) (High), 1 reference frame, vaapi_vld(left), 1280x720 [SAR 1:1 DAR 16:9], q=-1--1, 6000 kb/s, 59.94 fps, 59.94 tbn, 59.94 tbc
    Metadata:
      encoder         : Lavc58.54.100 h264_vaapi
*** 28 dup!
[MOS]:  CRITICAL - SubmitCommandBuffer:1008: Command buffer submission failed!
[CODEC]:CRITICAL - SubmitCommandBuffer:4072: MOS returned error, eStatus = 0x21
[CODEC]:CRITICAL - ExecuteSliceLevel:6993: MOS returned error, eStatus = 0x21
[CODEC]:CRITICAL - ExecuteEnc:4417: Slice level encoding failed.
[CODEC]:CRITICAL - Execute:586: MOS returned error, eStatus = 0x21
[LIBVA]:CRITICAL - EncodeInCodecHal:1148: DDI:Failed in Codechal!
[LIBVA]:CRITICAL - EndPicture:81: DDI:DdiEncode_EncodeInCodecHal return failure.
[h264_vaapi @ 0x4945cc0] Failed to end picture encode issue: 24 (internal encoding error).
[h264_vaapi @ 0x4945cc0] Encode failed: -5.
Video encoding failed
[AVIOContext @ 0x49467c0] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x4927fc0] Statistics: 6279312 bytes read, 2 seeks
Conversion failed!
```

tmm1, 2019-08-13:
If I try to use vaapi to decode mpeg2, there is a similar problem:

```
tmm1@nuc:~$ ./ffmpeg-4.2-ihd-master -noautoscale -t 3 -init_hw_device vaapi=intel:/dev/dri/renderD128 -hide_banner -loglevel verbose -hwaccel vaapi -hwaccel_output_format vaapi -i stream.mpg -c copy -filter_hw_device intel -vf "format=nv12|vaapi,hwupload" -b:v 6000k -maxrate 8000k -c:v h264_vaapi -level 42 -an -f mpegts -y /dev/null
[AVHWDeviceContext @ 0x477d440] libva: VA-API version 1.5.0
[AVHWDeviceContext @ 0x477d440] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x477d440] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x477d440] Initialised VAAPI connection: version 1.5
[AVHWDeviceContext @ 0x477d440] VAAPI driver: Intel iHD driver - 1.0.0.
[AVHWDeviceContext @ 0x477d440] Driver not found in known nonstandard list, using standard behaviour.
Input #0, mpegts, from 'stream.mpg':
  Duration: 00:00:30.39, start: 27664.978322, bitrate: 10381 kb/s
  Program 3100
    Stream #0:0[0x3e9]: Video: mpeg2video (Main), 1 reference frame ([2][0][0][0] / 0x0002), yuv420p(tv, progressive, left), 1280x720 [SAR 1:1 DAR 16:9], Closed Captions, 59.94 fps, 59.94 tbr, 90k tbn, 119.88 tbc
    Stream #0:1[0x3ea](eng): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, 5.1(side), fltp, 448 kb/s
    Stream #0:2[0x3eb](spa): Audio: ac3 ([129][0][0][0] / 0x0081), 48000 Hz, stereo, fltp, 192 kb/s (visual impaired) (descriptions)
Stream mapping:
  Stream #0:0 -> #0:0 (mpeg2video (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[mpegts @ 0x47b4a00] Correcting start time by 325534
[MOS]:  CRITICAL - SubmitCommandBuffer:1008: Command buffer submission failed!
[CODEC]:CRITICAL - SliceLevel:1178: MOS returned error, eStatus = 0x21
[CODEC]:CRITICAL - Execute:1295: Primitive level decoding failed.
[LIBVA]:CRITICAL - EndPicture:800: DDI:DdiDecode_DecodeInCodecHal return failure.
[mpeg2video @ 0x47bb840] Failed to end picture decode issue: 23 (internal decoding error).
[mpeg2video @ 0x47bb840] hardware accelerator failed to decode picture
Error while decoding stream #0:0: Input/output error
[MOS]:  CRITICAL - SubmitCommandBuffer:1008: Command buffer submission failed!
[CODEC]:CRITICAL - SliceLevel:1178: MOS returned error, eStatus = 0x21
[CODEC]:CRITICAL - Execute:1295: Primitive level decoding failed.
[LIBVA]:CRITICAL - EndPicture:800: DDI:DdiDecode_DecodeInCodecHal return failure.
[mpeg2video @ 0x47bb840] Failed to end picture decode issue: 23 (internal decoding error).
[mpeg2video @ 0x47bb840] hardware accelerator failed to decode picture
Error while decoding stream #0:0: Input/output error
[MOS]:  CRITICAL - SubmitCommandBuffer:1008: Command buffer submission failed!
[CODEC]:CRITICAL - SliceLevel:1178: MOS returned error, eStatus = 0x21
[CODEC]:CRITICAL - Execute:1295: Primitive level decoding failed.
[LIBVA]:CRITICAL - EndPicture:800: DDI:DdiDecode_DecodeInCodecHal return failure.
[mpeg2video @ 0x47bb840] Failed to end picture decode issue: 23 (internal decoding error).
[mpeg2video @ 0x47bb840] hardware accelerator failed to decode picture
Error while decoding stream #0:0: Input/output error
```

The same binaries work as expected on a different Broxton J3455 system with 4.4.59+ kernel.

wangyan-intel, 2020-05-13:
@tmm1 Is this issue existed still? If yes, I can take a look.

Xiaogangli-intel, 2020-09-11:
Hi @tmm1, could you update the libva version, and try again? Thank you.

XinfengZhang, 2021-01-31:
because @wangyan-intel  could not reproduce the issue, and no LIBVA_TRACE provided. I will close it, @tmm1 any concern or any additional information, please feel free to re-open it

closed 2021-01-31 by XinfengZhang

dalcombright, 2021-12-28:
I am running into this issue now while using an i7-11700k.

[hevc_vaapi @ 0x55cc823a8480] Failed to end picture encode issue: 24 (internal encoding error).
[hevc_vaapi @ 0x55cc823a8480] Encode failed: -5.

or I get this in most of my conversions:

[h264 @ 0x559bb0a4a740] Failed to end picture decode issue: 23 (internal decoding error).
[h264 @ 0x559bb0a4a740] hardware accelerator failed to decode picture

ffmpeg -hide_banner -loglevel info -init_hw_device vaapi=vaapi0:/dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -hwaccel_device vaapi0 -i /library/tv/[FILE].mkv -strict -2 -max_muxing_queue_size 2054 -filter_hw_device vaapi0 -vf format=nv12|vaapi,hwupload -map 0:v:0 -map 0:a:0 -map 0:s:0 -map 0:s:1 -c:v:0 hevc_vaapi -c:a:0 copy -c:s:0 copy -c:s:1 copy -y /tmp/unmanic/unmanic_file_conversion-1640504463.512407/[FILE]-WORKING-2.mkv

I wasn't able to find what release this was rolled into, but since its over 1 year since the fix maybe its something else?