intel/media-driver #709: [regression][ff3f0edd7eaf7a1d7afeabd62bad675f07106de6][gst-vaapi][avc dec]core dump when do decoding

wangzj0601, 2019-08-14:
platform: KBL
media-driver: ff3f0edd7eaf7a1d7afeabd62bad675f07106de6
gstreamer: 21798184a0588dbe0a423a03314022f3ad2d84be
gst-vaapi: 9e0c133a2403d5457de4e1e247f019a430516ac0

repo step
gst-launch-1.0 -vf filesrc location=./AUD_MW_E.264 ! h264parse ! vaapih264dec ! videoconvert dither=0 ! video/x-raw,format=I420 ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./test.yuv
Segmentation fault (core dumped)

uartie, 2019-08-15:
It looks like the bad commit got reverted in 4054fa6aca43fe0e61b5342c9a8fe228f6b356a0

But maintainers/commit failed to mention this issue.

wangzj0601, 2019-08-15:
Do anyone mention this issue, I will keep this issue one day, and will close it due to the bad commit got reverted in 4054fa6

closed 2019-08-16 by wangzj0601