intel/media-driver #723: [gsttreamer-vaapi][h265 encode] set different quality-factor but get same result in QVBR mode.

feiwan1, 2019-09-06:
When set different quality-factor values in QVBR bitrate-control mode, while the output encoded files get the same md5. 

Cmd:
$ gst-launch-1.0 filesrc location=honeybee640x480.nv12 ! videoparse format=nv12 width=640 height=480 ! vaapih265enc rate-control=qvbr quality-factor=10 bitrate=1500 target-percentage=100 ! filesink location=gst.qvbr.10.1500.h265

$ gst-launch-1.0 filesrc location=honeybee640x480.nv12 ! videoparse format=nv12 width=640 height=480 ! vaapih265enc rate-control=qvbr quality-factor=30 bitrate=1500 target-percentage=100 ! filesink location=gst.qvbr.30.1500.h265

$md5sum *.h265
1e95b9d1ebb3e54b1fe0f62e9ce36ea3  gst.qvbr.10.1000.h265
1e95b9d1ebb3e54b1fe0f62e9ce36ea3  gst.qvbr.30.1000.h265

Note: This need gstreamer-vaapi from:
https://gitlab.freedesktop.org/wangfei/gstreamer-vaapi hevc_brc branch.

uartie, 2020-03-11:
I'm pretty sure this happens because of #865

uartie, 2020-04-23:
Any progress on fixing this?

uartie, 2020-07-30:
@XinfengZhang can we get some progress here for HEVC case?  Also #379 may be related for AVC case.

uartie, 2020-07-30:
cc: @onabiull @Xiaogangli-intel

Xiaogangli-intel, 2021-09-22:
This issue should already be fixed, and there are no activities on it for more than 1 year. Close it, if you still meet this issue, please fill a new issue, thank you.

closed 2021-09-22 by Xiaogangli-intel