intel/media-driver #725: [ffmpeg][vaapi]hevc lp encode failed when GPU hang on ICL

fulinjie, 2019-09-09:
Root cause: default CTU size

After last sync, we came to a conclusion that default CTU should be adjust by driver.

And I verified that if we don't set the CTU size to 64 from ffmpeg level, the hang issue still happens.

ctx->slice_block_width = ctx->slice_block_height = 64;

related issue in jira:
https://jira.devtools.intel.com/browse/VIZ-15547

PR in ffmpeg to verify:
https://github.com/intel-media-ci/ffmpeg/pull/42

XinfengZhang, 2019-09-17:
maybe need a interface to query it, but it should be common instead of only for HEVC

XinfengZhang, 2020-03-10:
https://github.com/intel/libva/pull/385

Xiaogangli-intel, 2022-04-18:
This issue is fixed in latest driver. Please contact us if you still have concerns. Thank you.

closed 2022-04-18 by Xiaogangli-intel