intel/media-driver #726: [regression] conformance test failure for create surface with format VA_FOURCC_A2B10G10R10

uartie, 2019-09-11:
New failures [regressions] are introduced in libva conformance tests (libva-utils) with the following patch:

```
commit f8092716fe1b8d7e2d78f82111e26694f2366739
Author: furongzh <furong.zhang@intel.com>
Date:   Sat Aug 31 07:11:33 2019 +0800

    Add A2RGB10 and A2BGR10 in vaQuerySurfaceAttributes.
    
    Signed-off-by: furongzh <furong.zhang@intel.com>
```

The cases that fail include `CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib`.  The problem is that user can't create a surface with format VA_FOURCC_A2B10G10R10 which is a supported format returned by vaQuerySurfaceAttributes.

**The failure occurs in DdiMedia_OsFormatToMediaFormat because VA_FOURCC_A2B10G10R10 format is not handled in the switch statement.**

https://github.com/intel/libva-utils/pull/173 can help debug this issue.  Here is the result for one of the failures with scoped trace patch:

```
$ LIBVA_DRIVER_NAME=iHD test_va_api --gtest_filter=*CreateSurfacesNoConfigAttrib/136
Note: Google Test filter = *CreateSurfacesNoConfigAttrib/136
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from CreateSurfaces/VAAPICreateSurfaces
[ RUN      ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/136
libva info: VA-API version 1.6.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /home/uaeoff/Work/workspace/media/install/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
test_va_api_fixture.cpp:394: Failure
      Expected: VaapiStatus(expectation)
      Which is: VA_STATUS_SUCCESS
To be equal to: VaapiStatus(vaCreateSurfaces(m_vaDisplay, format, resolution.width, resolution.height, surfaces.data(), surfaces.size(), (attribs.size() != 0 ? const_cast<VASurfaceAttrib*>(attribs.data()) : __null), attribs.size()))
      Which is: VA_STATUS_ERROR_UNSUPPORTED_RT_FORMAT
Google Test trace:
test_va_api_createsurfaces.cpp:163: VASurfaceAttrib(type = VASurfaceAttribPixelFormat(1), flags = 3, value = VAGenericValueTypeInteger(1):808665665)
test_va_api_fixture.cpp:407: Failure
      Expected: VaapiStatus(0x00000000)
      Which is: VA_STATUS_SUCCESS
To be equal to: VaapiStatus(vaDestroySurfaces(m_vaDisplay, surfaces.data(), surfaces.size()))
      Which is: VA_STATUS_ERROR_INVALID_SURFACE
Google Test trace:
test_va_api_createsurfaces.cpp:163: VASurfaceAttrib(type = VASurfaceAttribPixelFormat(1), flags = 3, value = VAGenericValueTypeInteger(1):808665665)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/136, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 16x16) (8 ms)
[----------] 1 test from CreateSurfaces/VAAPICreateSurfaces (8 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (9 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/136, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 16x16)

1 FAILED TEST
```

FurongZhang, 2019-09-18:
https://github.com/intel/media-driver/pull/727

XinfengZhang, 2020-10-25:
the issue should already been fixed. because libva-utils conformance test is import to daily ci.  if there are still something wrong, please re-open it. thanks

closed 2020-10-25 by XinfengZhang