intel/media-driver #730: [gstreamer-vaapi][Jpeg encode] support 444 8bit encode for jpeg on ICL.

feiwan1, 2019-09-18:
Hi, 
The jpeg 444 8bit encode is a request from Iotg. So may need your help to support it in media-driver side.

gstreamer-vaapi need to be use:
https://gitlab.freedesktop.org/wangfei/gstreamer-vaapi.git jpeg444enc branch

cmd:
$ gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,format=Y444,width=1920,height=1080 ! vaapijpegenc !  filesink location=out.jpeg

Y444 in cmdline is same with 444P in media-driver.

This cmd will be fail for media-driver doesn't support some api:
1. QuerySurfaceAttributes doesn't support 444P format.
2. vaPutImage doesn't support 444P format.

For the 2 issue, I added some test code in my private tree of media-driver:
https://github.com/feiwan1/media-driver/commit/e4b95694305c7a4949134fd1a1c9e98bc455cdbb 

While, then the cmdline will be executed success, but the output jpeg files get some noise, not sure if any other place of media-driver need to be change/implemented.

The output jpeg after fix the 2 issue of media-driver:
[image]

uartie, 2019-09-18:
@feiwan1 did you try adding `jpegparse` after `vaapijpegdec` element?

uartie, 2019-09-18:
@feiwan1 also, can you add `-vf` option to `gst-launch-1.0` command line and post the output so we can see the negotiated caps?

feiwan1, 2019-09-18:
@uartie , after add jpegparse after vaapjpegenc, the output is same. And the caps should be fine, vaapijpegenc accept Y444 on its sink pad:

root@tlab:~/build/media/jpeg# gst-launch-1.0 -vf videotestsrc num-buffers=1 ! video/x-raw,format=Y444,width=1920,height=1080 ! vaapijpegenc ! filesink location=test.jpeg
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapiencodejpeg0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayX11\)\ vaapidisplayx11-1";
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)Y444, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)Y444, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVaapiEncodeJpeg:vaapiencodejpeg0.GstPad:sink: caps = video/x-raw, format=(string)Y444, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)Y444, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVaapiEncodeJpeg:vaapiencodejpeg0.GstPad:src: caps = image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFileSink:filesink0.GstPad:sink: caps = image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:00.000299920
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

uartie, 2019-09-18:
What are the caps output for `jpegparse`?

uartie, 2019-09-18:
On a side note: according to https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-encoding-input-format-and-max-resolution the AYUV format is support for JPEG encode input.  AYUV has same subsampling (4:4:4) as 444P, but still doesn't seem to work with gst-vaapi 😞

feiwan1, 2019-09-18:
Unfortunately, jpegparse shows YV12 (411 subsampling)in its output caps, but I don't think this format is correct, since in its supported caps formats, no 444 format defined. When using mediainfo or to decode the output with vaapijpegdec, it shows that the out jpeg is 444 subsampling. 

According my investigation to jpeg encode code in media-driver, seems 444P is mostly support, but AYUV is not indicated at all. @XinfengZhang @wangyan-intel , Is AYUV jpeg encode supported in media-driver? If yes, we can also use AYUV as input format since Iotg can accept both AYUV and 444P.

uartie, 2019-09-18:
Oh right, jpegparse doesn't report 444 correctly 😞

uartie, 2019-09-18:
Without any driver patches, it looks like ffmpeg-vaapi produces corrupted 444 jpeg too:
```
ffmpeg -v verbose -hwaccel vaapi -vaapi_device /dev/dri/renderD128 \
  -f lavfi -i smptebars -pix_fmt yuv444p -t 30 \
  -vf 'format=yuv444p,hwupload' \
  -c:v mjpeg_vaapi -y test.mjpeg
```

... so there must be somewhere else in the driver that is causing this problem.

feiwan1, 2019-10-16:
According the media-driver wiki, https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-encoding-input-format-and-max-resolution

Only AYUV but no Y444P support in media-dirver:
`
** JPEG encoding supports input format NV12/YUY2/UYVY/AYUV/ABGR/Y8
`
If so, we don't need to fix this issue, and only need to support AYUV format in jpeg encode. Thus, we will meet AYUV issue #742 report by @xhaihao .

feiwan1, 2019-10-16:
@XinfengZhang , please feel free to close this issue, if Y444P jpeg encode will not be supported.
Thanks.

closed 2019-11-11 by XinfengZhang