intel/media-driver #734: Rotation causes image colors to drift

uartie, 2019-09-25:
For 1280x720 resolutions (and possibly others), rotating 90 and/or 270 degrees causes some colors to drift from their original colors.

This issue can be reproduced with gstreamer-vaapi and ffmpeg-vaapi.  But can't be reproduced with gst-msdk or ffmpeg-qsv (i.e. msdk).

```
$ export LIBVA_DRIVER_NAME=iHD
$ export GST_VAAPI_ALL_DRIVERS=1

$ gst-launch-1.0 videotestsrc num-buffers=1 \
  ! video/x-raw,format=NV12,width=1280,height=720 \
  ! checksumsink2 frame-checksum=false dump-output=true dump-location=identity.yuv

$ gst-launch-1.0 -vf filesrc location=./identity.yuv \
  ! rawvideoparse format=nv12 width=1280 height=720 \
  ! vaapipostproc video-direction=90r ! video/x-raw,format=NV12 \
  ! vaapipostproc video-direction=90l ! video/x-raw,format=NV12 \
  ! checksumsink2 frame-checksum=false dump-output=true dump-location=gst-vaapi-rotation.yuv

$ gst-launch-1.0 -vf filesrc location=./identity.yuv \
  ! rawvideoparse format=nv12 width=1280 height=720 \
  ! msdkvpp video-direction=90r ! video/x-raw,format=NV12 \
  ! msdkvpp video-direction=90l ! video/x-raw,format=NV12 \
  ! checksumsink2 frame-checksum=false dump-output=true dump-location=gst-msdk-rotation.yuv

$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose \
  -f rawvideo -pix_fmt nv12 -s:v 1280x720 -i ./identity.yuv \
  -vf 'format=nv12|vaapi,hwupload,transpose_vaapi=dir=clock,transpose_vaapi=dir=cclock,hwdownload,format=nv12' \
  -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 1 -y ffmpeg-vaapi-rotation.yuv

$ ffmpeg -init_hw_device qsv=qsv:hw -hwaccel qsv -filter_hw_device qsv -v verbose \
  -f rawvideo -pix_fmt nv12 -s:v 1280x720 -i ./identity.yuv \
  -vf 'format=nv12|qsv,hwupload=extra_hw_frames=16,vpp_qsv=transpose=clock,vpp_qsv=transpose=cclock,hwdownload,format=nv12' \
  -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 1 -y ffmpeg-qsv-rotation.yuv

$ md5sum *.yuv

5dccaf4b7e2bfc0d27331e31064def53  ffmpeg-qsv-rotation.yuv
8f3e696da951061fe7194401dd178622  ffmpeg-vaapi-rotation.yuv
5dccaf4b7e2bfc0d27331e31064def53  gst-msdk-rotation.yuv
8f3e696da951061fe7194401dd178622  gst-vaapi-rotation.yuv
5dccaf4b7e2bfc0d27331e31064def53  identity.yuv
```

As seen in above results, gst-vaapi and ffmpeg-vaapi rotations don't match the identity.yuv, but msdk use-cases do.  Inspection of the mismatched yuv files reveal that some yuv color values are slightly changed.

Rotation should not cause colors to change.

This does not seem to be an issue with some resolution like 1920x1080.  Also, 180 degree rotation does not cause problems either.

Also see https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues/187 for other ways to replicate the issue.

uartie, 2020-02-05:
We are able to workaround this issue in gstreamer-vaapi (see https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/merge_requests/258).

It seems that this issue is related to setting input and output color standard to VAProcColorStandardNone.  And doing so causes other issues to manifest, too (https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues/226).

In the aforementioned gstreamer-vaapi patches (258), we changed input and output color standard to VAProcColorStandardBT709 and the issues disappear. 

Question is, what does the driver choose for input/output color standard when app sets to VAProcColorStandardNone?  And are those chosen standards sensible or naive?

AFAIK, gstreamer-vaapi and ffmpeg-vaapi have used the VAProcColorStandardNone setting for years without problems on other drivers like the one seen in 226 above.

uartie, 2020-02-24:
As mentioned in https://github.com/intel/media-driver/issues/760#issuecomment-590476083, the same commit can workaround this issue for ffmpeg-vaapi, too.

Alex1Zhang, 2020-02-25:
When app doesn't set color standard, this case will have two CSC leading to image colors drifting.
For 1280*720 driver will select BT.601 as the color space and for 720*1280 driver will select BT.709 as the color space after 90r. This is the first CSC. Vice versa. 
Thanks

XinfengZhang, 2020-03-05:
@uartie please help to check whether the gstreamer-vaapi set the color standard correctly?

uartie, 2020-03-05:
@XinfengZhang please read https://github.com/intel/media-driver/issues/734#issuecomment-582504001

uartie, 2020-03-05:
> When app doesn't set color standard, this case will have two CSC leading to image colors drifting.
> For 1280 x 720 driver will select BT.601 as the color space and for 720 x 1280 driver will select BT.709 as the color space after 90r. This is the first CSC. Vice versa.
> Thanks

But why would we want this behavior? I mean, they are essentially the same resolution, right?  Consider if a user is decoding a video on a cellphone or tablet, and they rotate the screen... (you can fill in the rest).

Am I being too naive or does this make sense?

uartie, 2020-03-05:
> When app doesn't set color standard, this case will have two CSC leading to image colors drifting.
> For 1280 x 720 driver will select BT.601 as the color space and for 720 x 1280 driver will select BT.709 as the color space after 90r. This is the first CSC. Vice versa.

I guess I don't understand why rotation (or scaling, for that matter) would imply we want to do CSC.

XinfengZhang, 2021-01-29:
@uartie , we already have fix in ffmpeg & gstreamer, also has fix in media driver as Furong mentioned in https://github.com/intel/media-driver/issues/760#issuecomment-607587378 . so I close the issue, if you still have concern, please feel free to re-open it, thanks

closed 2021-01-29 by XinfengZhang