intel/media-driver #736: Build fail under CentOS7.6 with its native gcc 4.8 compiler

HeFan2017, 2019-09-29:
Two building issues found by now:
1. In constructor 'PerfUtility::PerfUtility()', error: array used as initializer. Consider to use char string[num] = {'\0'} instead of char string[num] = ""
2. unrecognized command line option "-Wno-delete-incomplete" [-Werror]. It will cause the MOS_Delete(m_bltInterface) fail and mhwBltInterface not defined.

dvrogozh, 2019-09-30:
@HeFan2017 : media-driver is written with C++11. In gcc compiler official support for this standard appeared starting from gcc-6, before that support is experimental and it is not recommended by the media driver to be used: see very first note in https://github.com/intel/media-driver#known-issues-and-limitations. Moreover, I can saw for sure that with CentOS gcc-4.8 you _will_ step into compiler bug and meet weird issues running the driver, that known bug was addressed only in gcc-4.9.

If you need to build driver on CentOS, please, use one of the devtoolset coming with CentOS which gives higher compiler version support. I did that myself pretty long ago, but this should be kind of relevant instruction I hope: https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/

dvrogozh, 2019-10-09:
I won't take this issue for assignment since this is not an issue at all.

PR #740 (2020-01-16): Fix Build fail under CentOS7.6 with its native gcc 4.8 compiler
array used as initializer

fixes #736
closed 2020-01-16 by intel-mediadev