intel/media-driver #738: vaEndPicture fails if surface size is different from coded size

rosetta-jpn, 2019-10-02:
This is an upstream version about https://bugs.chromium.org/p/chromium/issues/detail?id=1009297#c16.

This happens with Chromium HW encoder stack using VA-API.
When this issue happens, the encoder is requested to encode 640x360 visible area stream.
Chromium HW encoder computes the coded size by aligning width and height both 16.
That said,
visible size: 640x360
coded size: 640x368 
coded size is computed by aligning width and height by 16.
On the other hand, Chromium HW encoder creates va surface with 640x360 because Chromium HW encoder aligns width by 64 and height by 4. [1, 2]

This situation causes a failure upon vaEndPicture(). No useful info is shown in
[libva trace log](https://github.com/intel/media-driver/files/3679178/libva_trace.txt)

Note that this issue doesn't happen on any chipsets with i965 driver.

The easy reproduce step is to use test framework called Tast in ChromiumOS project.
In chromiumos chroot, you should run
$ tast run $DUT_IP video.EncodeAccelH264360PI420

[1] https://cs.chromium.org/chromium/src/media/gpu/vaapi/vaapi_video_encode_accelerator.cc?l=382&rcl=1b8a6f47ba236a3043e1d796e24ae1ac4d3c094f
[2] https://chromium.googlesource.com/chromiumos/platform/minigbm/+/c062c3b8e12f9a99bd835acaa996602d920e531d/i915.c#239

rosetta-jpn, 2019-10-02:
Note that: this failure happens on VP8 encoding too.

XinfengZhang, 2019-10-09:
hi @rosetta-jpn , sorry for late response, just back from holiday.  you call vaCreateSurfaces to create the coded picture, right? is it external memory?  if you call vaCreateSurface to allocate frames, iHD will handle surface inside driver, it should create a surface with height=384, because if the surface is tile surface, it should align with 128x32.

XinfengZhang, 2020-10-25:
@rosetta-jpn no response, I will close it, if any other concern, please re-open it directly

closed 2020-10-25 by XinfengZhang