intel/media-driver #741: DdiVp_QueryVideoProcFilterCaps performs invalid memory write

umaplehurst, 2019-10-09:
Valgrind detected this problem when using media-driver with gStreamer, suggested patch follows, the capacity check should be made before writing to the destination:

```
==18145== Invalid write of size 4
==18145==    at 0xB85880D: DdiVp_QueryVideoProcFilterCaps(VADriverContext*, unsigned int, int, void*, unsigned int*) (media_libva_vp.c:3897)
==18145==    by 0x7F4B360: vpp_get_filter_caps_unlocked (gstvaapifilter.c:227)
==18145==    by 0x7F4B360: vpp_get_filter_caps (gstvaapifilter.c:261)
==18145==    by 0x7F4B360: get_operations_ordered (gstvaapifilter.c:671)
==18145==    by 0x7F4B360: get_operations (gstvaapifilter.c:718)
==18145==    by 0x7F23011: gst_vaapipostproc_ensure_filter_caps (gstvaapipostproc.c:268)
==18145==    by 0x7F23011: expand_allowed_srcpad_caps (gstvaapipostproc.c:1217)
==18145==    by 0x7F23011: ensure_allowed_srcpad_caps (gstvaapipostproc.c:1273)
==18145==    by 0x7F23011: gst_vaapipostproc_transform_caps_impl (gstvaapipostproc.c:1290)
==18145==    by 0x7F23011: gst_vaapipostproc_transform_caps (gstvaapipostproc.c:1307)
==18145==    by 0x3EC6E40CCD: gst_base_transform_transform_caps (gstbasetransform.c:465)
==18145==    by 0x3EC6E4287A: gst_base_transform_acceptcaps_default (gstbasetransform.c:1255)
==18145==    by 0x3EC6E459CF: gst_base_transform_default_query (gstbasetransform.c:1529)
==18145==    by 0x3EC527A35F: gst_pad_query (gstpad.c:4067)
==18145==    by 0x3EC52B4D4B: gst_pad_query_accept_caps (gstutils.c:3144)
==18145==    by 0x3EC5272A69: pre_eventfunc_check (gstpad.c:5607)
==18145==    by 0x3EC5272A69: gst_pad_send_event_unchecked (gstpad.c:5752)
==18145==    by 0x3EC52730A2: gst_pad_push_event_unchecked (gstpad.c:5406)
==18145==    by 0x3EC52734D8: push_sticky (gstpad.c:3922)
==18145==    by 0x3EC5270F87: events_foreach (gstpad.c:612)
```

```
diff --git a/media_driver/linux/common/vp/ddi/media_libva_vp.c b/media_driver/linux/common/vp/ddi/media_libva_vp.c
index 84d09540..5c72d4d1 100644
--- a/media_driver/linux/common/vp/ddi/media_libva_vp.c
+++ b/media_driver/linux/common/vp/ddi/media_libva_vp.c
@@ -3893,13 +3893,13 @@ DdiVp_QueryVideoProcFilterCaps (
             {
                 VAProcFilterCapDeinterlacing* diCap = (VAProcFilterCapDeinterlacing*) filter_caps;
 
-                diCap[0].type                         = VAProcDeinterlacingBob;
-                diCap[1].type                         = VAProcDeinterlacingMotionAdaptive;
-
                 if (uQueryCapsNum < uExistCapsNum)
                 {
                     return VA_STATUS_ERROR_MAX_NUM_EXCEEDED;
                 }
+
+                diCap[0].type                         = VAProcDeinterlacingBob;
+                diCap[1].type                         = VAProcDeinterlacingMotionAdaptive;
             }
             break;
 
@@ -3936,6 +3936,11 @@ DdiVp_QueryVideoProcFilterCaps (
             {
                 VAProcFilterCapColorBalance* ColorBalCap;
 
+                if (uQueryCapsNum < uExistCapsNum)
+                {
+                    return VA_STATUS_ERROR_MAX_NUM_EXCEEDED;
+                }
+
                 for (uCnt=0 ; uCnt<uQueryCapsNum ; uCnt++)
                 {
                     /* the actual filter caps attribute is not enough */
@@ -3951,11 +3956,6 @@ DdiVp_QueryVideoProcFilterCaps (
                     ColorBalCap->range.min_value        = VpColorBalCap[uCnt].range.min_value;
                     ColorBalCap->range.step             = VpColorBalCap[uCnt].range.step;
                 }
-
-                if (uQueryCapsNum < uExistCapsNum)
-                {
-                    return VA_STATUS_ERROR_MAX_NUM_EXCEEDED;
-                }
             }
             break;
         case VAProcFilterSkinToneEnhancement:
@@ -3978,6 +3978,9 @@ DdiVp_QueryVideoProcFilterCaps (
             {
                 VAProcFilterCapTotalColorCorrection* TccCap;
 
+                if (uQueryCapsNum < uExistCapsNum)
+                    return VA_STATUS_ERROR_MAX_NUM_EXCEEDED;
+
                 for (uCnt = 0; uCnt < uQueryCapsNum; uCnt++)
                 {
                     if (uCnt >= uExistCapsNum)
@@ -3990,9 +3993,6 @@ DdiVp_QueryVideoProcFilterCaps (
                     TccCap->range.default_value = TCC_DEFAULT;
                     TccCap->range.step          = TCC_STEP;
                 }
-
-                if (uQueryCapsNum < uExistCapsNum)
-                    return VA_STATUS_ERROR_MAX_NUM_EXCEEDED;
             }
             break;
         case VAProcFilterHighDynamicRangeToneMapping:

```

uartie, 2019-10-09:
@umaplehurst please check #728.

umaplehurst, 2019-10-09:
Thank you, I'll close this one.

closed 2019-10-09 by umaplehurst