intel/media-driver #742: JPEG encoder doesn't support AYUV input

xhaihao, 2019-10-10:
According to https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-encoding-input-format-and-max-resolution, JPEG encoder supports AYUV input, however `DdiEncodeJpeg::ConvertMediaFormatToInputSurfaceFormat` doesn't convert AYUV to a valid internal format.

https://github.com/intel/media-driver/blob/cf6e2ca57cbd7b5c1aa743807d3a5cac8202cacb/media_driver/linux/common/codec/ddi/media_ddi_encode_jpeg.cpp#L582-L606

XinfengZhang, 2019-12-09:
AYUV should be similar with RGB and 444P

feiwan1, 2019-12-11:
> AYUV should be similar with RGB and 444P

Hi @XinfengZhang , so you will plan to add AYUV format in jpeg encode in media-driver?

Xiaogangli-intel, 2022-04-24:
JPEG encoder doesn't support AYUV format. Feature table is updated in #1388.

closed 2022-04-24 by Xiaogangli-intel