intel/media-driver #744: DdiMediaBase::GetRenderTargetID cost 2% cpu time when we do single process multiple channel decoder

xuguangxin, 2019-10-10:
If you use vtune to profile following command
`gst-launch-1.0 tee name = t filesrc location=1920x1080_2700frames.264 ! t. t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false t. ! h264parse ! queue ! vaapih264dec ! "video/x-raw(memory:VASurface)" ! fpsdisplaysink video-sink=fakesink sync=false`

you will see DdiMediaBase::GetRenderTargetID cost about 2% cpu time.
[image]
This becuase gstreamer vaapi did not fill the entire reference list.

https://github.com/GStreamer/gstreamer-vaapi/blob/600aba57cfe91ea991c41f75f7ad347621ec4f5a/gst-libs/gst/vaapi/gstvaapidecoder_h264.c#L4171

But the driver will blindly fill entire reference list 

https://github.com/intel/media-driver/blob/cf6e2ca57cbd7b5c1aa743807d3a5cac8202cacb/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp#L116

Every surface id beyond num_ref_idx_l0_active_minus1  is invalid surface id. 
and DdiMediaBase::GetRenderTargetID will loop DDI_MEDIA_MAX_SURFACE_NUMBER_CONTEXT(127) times for an invalid id.
We support 2x32 reference frames for each slice. In the worst case, it will loop 2x32*127 time for each slice.
Please help change it.

xuguangxin, 2019-10-24:
@XinfengZhang any eta for this?

xuguangxin, 2019-11-06:
@XinfengZhang @wangyan42164, any feedback for this?
thanks

wangyan-intel, 2020-06-28:
@xuguangxin Could you please try my PR: #986? Thanks

xuguangxin, 2020-06-29:
@wangyan-intel In which condition we will return here?
https://github.com/intel/media-driver/blob/adc23261d4a4f60d6e113959cdb3acbed055fde9/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp#L832
In this condition,  we still loop the entire array, will it have problem?

thanks

wangyan-intel, 2020-06-30:
> @wangyan-intel In which condition we will return here?
> https://github.com/intel/media-driver/blob/adc23261d4a4f60d6e113959cdb3acbed055fde9/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp#L832
> 
> 
> In this condition, we still loop the entire array, will it have problem?
> thanks

@xuguangxin IMHO, we don't know how many frames in avcPicParams->RefFrameList. So we have to search all elements for this slice. Do you think so?
https://github.com/intel/media-driver/blob/adc23261d4a4f60d6e113959cdb3acbed055fde9/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp#L127
https://github.com/intel/media-driver/blob/adc23261d4a4f60d6e113959cdb3acbed055fde9/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp#L141

xuguangxin, 2020-06-30:
Array is not good for search. 
Is it possible use some std::map instead of Array.
else if platform supports compressed memory, we still have this high overhead.

wangyan42164, 2020-07-02:
@xuguangxin 
Yes. It could be changed tree structure like std:map for searching. But it need more effort and test for avoid regression.

xuguangxin, 2020-07-06:
@wangyan42164 let us merge your patch firstly. then we will measure the performance at TGL, create a new issues if we still see the problem.
thanks

wangyan-intel, 2020-07-06:
@xuguangxin Sure. I have added verifying flag for #986.

PR #986 (2020-07-09): Avoid search invalid ref frames/fields.
Fixes #744.
commit 7249f5c0 (2020-07-09): Avoid search invalid ref frames/fields when AVC decoding

Fixes #744.
closed 2020-07-09 by intel-mediadev