intel/media-driver #745: Bad perf & lower GPU utilization with other than CQP bitrate control in low-power mode

eero-t, 2019-10-11:
Setup:
- HW: GEN9+
- OS: Ubuntu 18.04
- SW: drm-tip kernel and rest of media stack built from git
- HuC loaded to use other bit-rate controls, e.g. by having following in "/etc/modprobe.d/i915.conf": `options i915 enable_guc=2`
- "sudo intel_gpu_top" from "intel-gpu-tools" project running on console, for monitoring GPU utilization

Use-case:
* Low-power AVC transcoding with bit-rate controls supported by the MSDK sample application:
```
for control in la cbr cqp vbr; do
  sample_multi_transcode -i::h264 720x480p_30.00_4mb_h264_cabac_180s.264 -o::h264 0038_SD03_1.0.h264 -b 2000 -u 4 -async 4 -hw -qsv-ff -$control;
done
```

Actual outcome:
* CQP 50% faster than the other bit-rate controls
* GPU fully utilized only with CQP, in other cases utilization is just ~85% (GPU runs at full speed though)

I see the performance difference regardless of which GEN9+ HW I use;  BXT, SKL, KBL, ICL...

Although MSDK documentation states CQP to be fastest:
* https://software.intel.com/en-us/articles/common-bitrate-control-methods-in-intel-media-sdk

This perf gap is really significant, and with the other bit-rate controls utilizing GPU only partially, it would seem that there's room to improve also their performance.

Differences are smaller in other AVC cases with larger input resolutions, but still very noticeable (tens of percents).  In transcode operation where output is scaled to 352x240 (along with frame rate conversion), there's no difference visible.

XinfengZhang, 2020-07-08:
@eero-t , other bitrate control has BRC algorithm involved, why expect same perf?

eero-t, 2020-07-13:
> @eero-t , other bitrate control has BRC algorithm involved, why expect same perf?

"...with the other bit-rate controls utilizing GPU only partially, it would seem that there's room to improve also their performance."

Or do you mean that BRC requires more syncing with CPU side, which is unavoidable?

XinfengZhang, 2020-07-14:
BRC is handled by Huc, @Xiaogangli-intel , do you have the data whether HUC BRC will impact perf? and how much impact?

Xiaogangli-intel, 2020-09-11:
@eero-t, some optimizations were checked-in, could you re-run the test and check whether the issue is gone.

eero-t, 2020-09-14:
When using Media stack (drm-tip kernel, gmmlib, libva, media-driver, mediasdk) built from yesterday Git heads, and testing on CML-H...

According to "intel_gpu_top", CQP has, when compared to to other rate-controls:
* 50% higher FPS
* 5% higher power consumption
* 100% GPU utilization instead of 95%

I.e. there's still the same 2x performance gap (which seems still too high, when considering that the other rate-controls don't use GPU fully).

PS. I haven't seen any noticeable (GEN9) Media performance increase in past 1/2 year, except for one in early August which fixes end of March performance regression on Atoms.

Xiaogangli-intel, 2021-09-22:
There are no activities on this issue for almost 1 year, I will close it. If you still can meet this issue, please file a new one for it. Thank you.

closed 2021-09-22 by Xiaogangli-intel