intel/media-driver #746: userptr: gmmlib guesses end-user memory allocation layout

dvrogozh, 2019-10-11:
Media driver calls gmmlib to get gmm resource object here:
https://github.com/intel/media-driver/blob/fda5ff183099a20ca3541d67b7260f3ca0f12fa4/media_driver/linux/common/ddi/media_libva_util.cpp#L403

In case of userptr path this call should check memory allocation done by the end user and return corresponding gmm resource object or `NULL` if allocation can't be accepted. However, media driver does not fully describe memory allocation layout to the gmmlib which is hinted by this call below:
https://github.com/intel/media-driver/blob/fda5ff183099a20ca3541d67b7260f3ca0f12fa4/media_driver/linux/common/ddi/media_libva_util.cpp#L418

I.e. media driver does not pass pitch initially to the gmmlib which means that gmm just guess  allocation parameters which can lead to unexpected behavior: denial of service is one of the possible outcomes.

The reason why media driver did not pass pitch to gmmlib is lack of required uAPI. I will submit feature request to gmm.

XinfengZhang, 2021-01-31:
the issue should already been fixed by https://github.com/intel/media-driver/commit/c1ce9538718c8b775bb2bd3be7011757d59a3592 &  https://github.com/intel/gmmlib/commit/1598c1b46fb911cb9060d1a5380aec1ca1ef2c19

Jexu, 2022-12-02:
Could you help confirm it since it may have been fixed by above pr.

dvrogozh, 2022-12-03:
That's a complex task for which I don't have a bandwidth. If you believe it's closed - close it.

closed 2024-07-05 by MicroYY