intel/media-driver #749: Build failed on Ubuntu Disco 'intel-media-driver-644fc6d2bb9e'

zhoujd, 2019-10-12:
our CI found following commit id : 
644fc6d2bb9e6271a2b91578b1bc2f63275d184a cause build failed on Ubuntu disco.

07:38:05,237 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h: In function 'int parse_separate_sysfs_files(int, int, drmPciDeviceInfoPtr, bool)':
07:38:05,237 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:535:46: error: '/driver' directive output may be truncated writing 7 bytes into a region of size between 1 and 4097 [-Werror=format-truncation=]
07:38:05,237 INFO  - snprintf(driverpath, sizeof(driverpath), "%s/driver", pci_path);

https://media-ci.ostc.intel.com/build/340127
It is blocking our CI.
Please let me know if you need more information.

XinfengZhang, 2019-10-14:
@HeFan2017  please help to check it

HeFan2017, 2019-10-14:
Sure, we will check this issue.

xhaihao, 2019-10-14:
I also have this issue, could you please fix it  or revert the bad commit ASAP?

XinfengZhang, 2019-10-15:
already reverted by 3a039596739b548b4fb20ce7bbbfd431122cc457

closed 2020-07-08 by XinfengZhang