intel/media-driver #754: Devult failed with Clang-8

egorovdanil, 2019-10-17:
Running devult...
[==========] Running 43 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 1 test from MediaCapsDdiTest
[ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [25]
param: 4, val: 0
/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:99: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 18
Platform = SKL, Failed function = m_driverLoader.InitDriver

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:103: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: -1
Platform = SKL, Failed function = Test_QueryConfigProfiles

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:106: Failure
Value of: (CompareFeatureIDTable(queriedFeatureIDTable, refFeatureIDTable))
  Actual: false
Expected: true
Platform = SKL, Failed function = CompareFeatureIDTable

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:110: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.CloseDriver

DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [25]
param: 4, val: 0
/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:99: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 18
Platform = BXT, Failed function = m_driverLoader.InitDriver

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:103: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: -1
Platform = BXT, Failed function = Test_QueryConfigProfiles

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:106: Failure
Value of: (CompareFeatureIDTable(queriedFeatureIDTable, refFeatureIDTable))
  Actual: false
Expected: true
Platform = BXT, Failed function = CompareFeatureIDTable

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:110: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = BXT, Failed function = m_driverLoader.CloseDriver

DRM_IOCTL_I915_GEM_APERTURE failed: Bad file descriptor
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [9]
param: 4, val: 0
/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:99: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 18
Platform = BDW, Failed function = m_driverLoader.InitDriver

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:103: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: -1
Platform = BDW, Failed function = Test_QueryConfigProfiles

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:106: Failure
Value of: (CompareFeatureIDTable(queriedFeatureIDTable, refFeatureIDTable))
  Actual: false
Expected: true
Platform = BDW, Failed function = CompareFeatureIDTable

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:110: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = BDW, Failed function = m_driverLoader.CloseDriver

[  FAILED  ] MediaCapsDdiTest.DecodeEncodeProfile (4 ms)
[----------] 1 test from MediaCapsDdiTest (4 ms total)

[----------] 2 tests from MediaDecodeDdiTest
[ RUN      ] MediaDecodeDdiTest.DecodeHEVCLong
DRM_IOCTL_I915_GEM_APERTURE failed: Inappropriate ioctl for device
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [25]
param: 4, val: 0
/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:65: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 18
Platform = SKL, Failed function = m_driverLoader.InitDriver

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:72: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateConfig

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:78: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateSurfaces2

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:83: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateContext

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:90: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaBeginPicture

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:98: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaCreateBuffer

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

/home/a/MSDKOpenSource/media-driver/media_driver/linux/ult/ult_app/ddi_test_decode.cpp:108: Failure
      Expected: 0x00000000
      Which is: 0
To be equal to: ret
      Which is: 5
Platform = SKL, Failed function = m_driverLoader.m_ctx.vtable->vaRenderPicture

Segmentation fault (core dumped)
media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:62: recipe for target 'RunULT' failed
make[2]: *** [RunULT] Error 139
CMakeFiles/Makefile2:374: recipe for target 'media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all' failed
make[1]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

XinfengZhang, 2019-11-05:
it should already been fixed by https://github.com/intel/media-driver/commit/0e840fa129f3ce9f46e2132784986e5f47dc86a3 , one function should be exported

XinfengZhang, 2019-11-05:
@egorovdanil could you help to double confirm?

egorovdanil, 2019-11-15:
Thank you, problem solved.

closed 2019-11-15 by egorovdanil