intel/media-driver #755: Build failed with Clang-8 due to -Werror

egorovdanil, 2019-10-18:
```
clang: error: -Wl,--gc-sections: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: -Wl,--no-as-needed: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: optimization flag '-funswitch-loops' is not supported [-Werror,-Wignored-optimization-argument]
clang: error: optimization flag '-finline-limit=100' is not supported [-Werror,-Wignored-optimization-argument]
clang: error: -Wl,--gc-sections: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: -Wl,--no-as-needed: 'linker' input unused [-Werror,-Wunused-command-line-argument]
clang: error: optimization flag '-funswitch-loops' is not supported [-Werror,-Wignored-optimization-argument]
clang: error: optimization flag '-finline-limit=100' is not supported [-Werror,-Wignored-optimization-argument]
```

XinfengZhang, 2019-11-05:
@egorovdanil  could you help to check https://github.com/intel/media-driver/commit/0e840fa129f3ce9f46e2132784986e5f47dc86a3

egorovdanil, 2019-11-15:
Thank you, problem solved.

closed 2019-11-15 by egorovdanil