intel/media-driver #757: Memory corruption / segfault on yuyv422 hwdownload

eero-t, 2019-10-18:
Setup:
* HW: KBL-i7 GT3e
* OS: Ubuntu 18.04
* SW: Git versions of drm-tip kernel (v5.2 or newer) and media stack

Between following commits from 2019-07-19:
* 87125b3557 Avoid invalid Max QP when AVC encoding.
* 443241921c [VP] Add tile type for image creation

iHD media-driver had started to crash when doing uyvy422 format hwdownload.  Before that FFmpeg conversion worked without issues.

Following outputs are with yesterday git version:
```
$ ffmpeg -y -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i 720x480p_30.00_4mb_h264_cabac_180s.264 -c:v rawvideo -vf hwdownload,format=uyvy422 -f rawvideo /dev/null
...
Input #0, h264, from '/home/testrunner/Videos/queue/720x480p_30.00_4mb_h264_cabac_180s.264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(tv, smpte170m, progressive), 720x480 [SAR 10:11 DAR 15:11], 30 fps, 30 tbr, 1200k tbn, 60 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
Output #0, rawvideo, to '/dev/null':
  Metadata:
    encoder         : Lavf58.32.104
    Stream #0:0: Video: rawvideo (UYVY / 0x59565955), uyvy422, 720x480 [SAR 10:11 DAR 15:11], q=2-31, 165888 kb/s, 30 fps, 30 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.56.102 rawvideo
corrupted double-linked list
Aborted
```

When trying the same with videos having larger resolution, FFmpeg segfaults.

According to Gdb, abort happens here:
```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff485a801 in __GI_abort () at abort.c:79
#2  0x00007ffff48a3897 in __libc_message (action=action@entry=do_abort, 
    fmt=fmt@entry=0x7ffff49d0b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff48aa90a in malloc_printerr (str=str@entry=0x7ffff49cecba "corrupted double-linked list")
    at malloc.c:5350
#4  0x00007ffff48b213a in _int_free (have_lock=0, p=<optimized out>, av=0x7ffff4c05c40 <main_arena>)
    at malloc.c:4295
#5  __GI___libc_free (mem=<optimized out>) at malloc.c:3124
#6  0x00007fffedd98a2d in DdiMediaUtil_FreeSurface (surface=surface@entry=0x555557413ee0)
    at media_driver/linux/common/ddi/media_libva_util.cpp:984
#7  0x00007fffedd83cc4 in DdiMedia_DestroySurfaces (ctx=ctx@entry=0x5555573979c0, 
    surfaces=surfaces@entry=0x7fffffffd7a0, num_surfaces=num_surfaces@entry=1)
    at media_driver/linux/common/ddi/media_libva.cpp:1983
#8  0x00007fffedd89bfd in DdiMedia_DestroySurfaces (num_surfaces=1, surfaces=0x7fffffffd7a0, 
    ctx=0x5555573979c0)
    at media_driver/linux/common/ddi/media_libva.cpp:4645
#9  DdiMedia_GetImage (ctx=0x5555573979c0, surface=<optimized out>, x=<optimized out>, 
    y=<optimized out>, width=<optimized out>, height=<optimized out>, image=0)
    at media_driver/linux/common/ddi/media_libva.cpp:4647
#10 0x00007ffff625a36d in vaGetImage (dpy=0x555557397850, surface=23, x=0, y=0, width=720, height=480, 
    image=0) at va/va.c:1753
```
Segfault looks like this:
```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/home/testrunner/Videos/queue/bbb_sunflower_2160p_30fps_10sec.mp4':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: mp42mp41
    creation_time   : 2018-08-01T08:23:02.000000Z
  Duration: 00:00:10.05, start: 0.000000, bitrate: 30281 kb/s
    Stream #0:0(eng): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 3840x2160, 30094 kb/s, 30 fps, 30 tbr, 30k tbn, 60 tbc (default)
...
Thread 1 "ffmpeg" received signal SIGSEGV, Segmentation fault.
0x00007fffed5e62b1 in GmmLib::GmmClientContext::DestroyResInfoObject (this=0x5555573a61a0, 
    pResInfo=0x55555a496680)
    at Source/GmmLib/GlobalInfo/GmmClientContext.cpp:460
460	Source/GmmLib/GlobalInfo/GmmClientContext.cpp: No such file or directory.
(gdb) bt
#0  0x00007fffed5e62b1 in GmmLib::GmmClientContext::DestroyResInfoObject (this=0x5555573a61a0, 
    pResInfo=0x55555a496680)
    at Source/GmmLib/GlobalInfo/GmmClientContext.cpp:460
#1  0x00007fffedd98a2d in DdiMediaUtil_FreeSurface (surface=surface@entry=0x555557407b00)
    at media_driver/linux/common/ddi/media_libva_util.cpp:984
#2  0x00007fffedd83cc4 in DdiMedia_DestroySurfaces (ctx=ctx@entry=0x5555573979c0, 
    surfaces=surfaces@entry=0x7fffffffd7a0, num_surfaces=num_surfaces@entry=1)
    at media_driver/linux/common/ddi/media_libva.cpp:1983
#3  0x00007fffedd89bfd in DdiMedia_DestroySurfaces (num_surfaces=1, surfaces=0x7fffffffd7a0, 
    ctx=0x5555573979c0)
    at media_driver/linux/common/ddi/media_libva.cpp:4645
#4  DdiMedia_GetImage (ctx=0x5555573979c0, surface=<optimized out>, x=<optimized out>, 
    y=<optimized out>, width=<optimized out>, height=<optimized out>, image=0)
    at media_driver/linux/common/ddi/media_libva.cpp:4647
#5  0x00007ffff625a36d in vaGetImage (dpy=0x555557397850, surface=22, x=0, y=0, width=3840, 
    height=2160, image=0) at va/va.c:1753
```

eero-t, 2019-10-18:
Note: yuyv422 format works fine, only uyvy422 fails.

XinfengZhang, 2019-11-05:
@Jister , could you help to check, the issue should exist in vaGetImage, just the output format is different

Jister, 2019-11-19:
Could you try the patch #778?

eero-t, 2019-11-20:
**Test-case finishes fine with the patch**.  Without it, there's still a crash (at least with yesterday evening version).

Note, that while patch built fine for me:
https://github.com/intel/media-driver/commit/e87e34541a972a0e5fc10f9da53713600443b61d.patch

According to travis CI log: https://travis-ci.org/intel/media-driver/builds/613809709

Its build fails due to missing include file (path?):
```
/opt/src/media-driver/media_driver/agnostic/gen11_icllp/media_interfaces/media_interfaces_g11_ehl.cpp:31:10: fatal error: igcodeckrn_g11.h: No such file or directory

 #include "igcodeckrn_g11.h"

          ^~~~~~~~~~~~~~~~~~

compilation terminated.
```
Is there some issue in the media-driver travis CI setup?

(I see that include file in "media_driver/agnostic/gen11/codec/kernel_free/igcodeckrn_g11.h".)

PR #778 (2019-11-27): [VP] Change image tile type for UYVY
Align the tile type of image for UYVY/YVYU/VYUY with surface tile type

fix #757
closed 2019-11-27 by intel-mediadev