intel/media-driver #759: ffmpeg use vaapi system reboot

ailumiyana, 2019-10-22:
OS: Ubuntu 18.04 64bit, kernel version: 4.15.0
CPU: Intel i7-9700k

libva info: VA-API version 1.6.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.6 (libva 2.5.0)
vainfo: Driver version: Intel iHD driver - 1.0.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD

sudo ffmpeg -protocol_whitelist "file,http,https,rtp,udp,tcp,tls" -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i temp.sdp -lavfi "[0:v] hwupload, scale_vaapi=w=960:h=540:format=nv12,hwdownload,format=nv12 [s0];color=black:r=30:size=1920x1080:sar=1/1 [b0];[b0][s0] overlay=x=0:y=0, format=nv12|vaapi, hwupload" -c:v h264_vaapi out.264

ffmpeg version N-94335-g9869e21776 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --enable-hwaccel=h264_vaapi --enable-gpl --enable-decoder=h264 --disable-x86asm --enable-shared --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib
  libavutil      56. 30.100 / 56. 30.100
  libavcodec     58. 53.101 / 58. 53.101
  libavformat    58. 28.102 / 58. 28.102
  libavdevice    58.  7.100 / 58.  7.100
  libavfilter     7. 56.101 /  7. 56.101
  libswscale      5.  4.101 /  5.  4.101
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
[h264 @ 0x55d12401dec0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55d12401dec0] decode_slice_header error
[h264 @ 0x55d12401dec0] no frame!
[h264 @ 0x55d12401dec0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55d12401dec0] decode_slice_header error
[h264 @ 0x55d12401dec0] no frame!
[h264 @ 0x55d12401dec0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55d12401dec0] decode_slice_header error
[h264 @ 0x55d12401dec0] no frame!
[h264 @ 0x55d12401dec0] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55d12401dec0] decode_slice_header error
[h264 @ 0x55d12401dec0] no frame!
Input #0, sdp, from 'temp.sdp':
  Duration: N/A, start: 0.029000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1280x720, 29.97 tbr, 90k tbn, 180k tbc
File 'out.264' already exists. Overwrite ? [y/N] y
Stream mapping:
  Stream #0:0 (h264) -> hwupload
  hwupload -> Stream #0:0 (h264_vaapi)
Press [q] to stop, [?] for help
Duplicate formats in ff_merge_formats detected
[h264_vaapi @ 0x55d12407a8c0] No quality level set; using default (20).
Output #0, h264, to 'out.264':
  Metadata:
    encoder         : Lavf58.28.102
    Stream #0:0: Video: h264 (h264_vaapi) (High), vaapi_vld, 1920x1080 [SAR 1:1 DAR 16:9], q=-1--1, 30 fps, 30 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.53.101 h264_vaapi
[sdp @ 0x55d124017180] max delay reached. need to consume packet
[sdp @ 0x55d124017180] RTP: missed 112 packets
[h264 @ 0x55d1242b18c0] Increasing reorder buffer to 1
[sdp @ 0x55d124017180] max delay reached. need to consume packetate=3117.9kbits/s speed=   1x    
[sdp @ 0x55d124017180] RTP: missed 2 packets
[h264 @ 0x55d1242b4880] error while decoding MB 73 42, bytestream -20
[h264 @ 0x55d1242b4880] concealing 216 DC, 216 AC, 216 MV errors in I frame
temp.sdp: corrupt decoded frame in stream 0
[sdp @ 0x55d124017180] max delay reached. need to consume packetate=3114.6kbits/s speed=   1x    
[sdp @ 0x55d124017180] RTP: missed 10 packets
[h264 @ 0x55d1242b4880] error while decoding MB 69 37, bytestream -30
[h264 @ 0x55d1242b4880] concealing 620 DC, 620 AC, 620 MV errors in I frame
temp.sdp: corrupt decoded frame in stream 0
[sdp @ 0x55d124017180] max delay reached. need to consume packetrate=3290.3kbits/s speed=   1x    
[sdp @ 0x55d124017180] RTP: missed 10 packets
[h264 @ 0x55d1241e9d40] error while decoding MB 14 38, bytestream -24
[h264 @ 0x55d1241e9d40] concealing 595 DC, 595 AC, 595 MV errors in I frame
[sdp @ 0x55d124017180] max delay reached. need to consume packetrate=3289.5kbits/s speed=   1x    
[sdp @ 0x55d124017180] RTP: missed 15 packets
[h264 @ 0x55d1243359c0] error while decoding MB 14 35, bytestream -43
[h264 @ 0x55d1243359c0] concealing 835 DC, 835 AC, 835 MV errors in I frame
temp.sdp: corrupt decoded frame in stream 0
frame=157418 fps= 30 q=-0.0 size= 2172672kB time=01:27:27.20 bitrate=3392.0kbits/s speed=   1x    
Socket error Event: 32 Error: 10053.
Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(192.168.0.77) at 16:36:06.

Type `help' to learn how to use Xshell prompt.
[E:\~]$ 

here my ssh link down，system reboot.

top info : 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                             
 3105 root      20   0  895656 114840  37572 S  70.3   0.7  60:26.63 ffmpeg                                                                                              
   10 root      20   0       0      0      0 I   0.3   0.0   0:16.14 rcu_sched                                                                                           
   13 root      20   0       0      0      0 I   0.3   0.0   0:01.15 kworker/0:1-events 

Could anyone give some advice?

ailumiyana, 2019-10-22:
Another log :

sudo ffmpeg -i onlyI1080p.h264 -vaapi_device /dev/dri/renderD128 -lavfi "scale=w=1280x720,format=nv12|vaapi, hwupload" -c:v h264_vaapi -threads 1 out.264
[sudo] password for sola: 
ffmpeg version N-94335-g9869e21776 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --enable-hwaccel=h264_vaapi --enable-gpl --enable-decoder=h264 --disable-x86asm --enable-shared --enable-libmfx --enable-encoder=h264_qsv --enable-decoder=h264_qsv --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib
  libavutil      56. 30.100 / 56. 30.100
  libavcodec     58. 53.101 / 58. 53.101
  libavformat    58. 28.102 / 58. 28.102
  libavdevice    58.  7.100 / 58.  7.100
  libavfilter     7. 56.101 /  7. 56.101
  libswscale      5.  4.101 /  5.  4.101
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
Input #0, h264, from 'onlyI1080p.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 1200k tbn, 60 tbc
File 'out.264' already exists. Overwrite ? [y/N] y
Stream mapping:
  Stream #0:0 (h264) -> scale
  hwupload -> Stream #0:0 (h264_vaapi)
Press [q] to stop, [?] for help
[h264_vaapi @ 0x56156d0ee2c0] No quality level set; using default (20).
Output #0, h264, to 'out.264':
  Metadata:
    encoder         : Lavf58.28.102
    Stream #0:0: Video: h264 (h264_vaapi) (High), vaapi_vld, 1280x720 [SAR 1:1 DAR 16:9], q=-1--1, 30 fps, 30 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.53.101 h264_vaapi
frame=   73 fps=0.0 q=-0.0 Lsize=    2365kB time=00:00:02.40 bitrate=8072.3kbits/s speed=2.59x    
video:2365kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Exiting normally, received signal 2.
sola@vision-server:~/old_backup/home/sola/server_source/temp$ sudo ffmpeg -protocol_whitelist "file,http,https,rtp,udp,tcp,tls" -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i temp.sdp -lavfi "[0:v] hwupload, scale_vaapi=w=960:h=540:format=nv12,hwdownload,format=nv12 [s0];color=black:r=30:size=1920x1080:sar=1/1 [b0];[b0][s0] overlay=x=0:y=0, format=nv12|vaapi, hwupload" -c:v h264_vaapi out.264
[sudo] password for sola: 
ffmpeg version N-94335-g9869e21776 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 8 (Ubuntu 8.3.0-6ubuntu1)
  configuration: --enable-hwaccel=h264_vaapi --enable-gpl --enable-decoder=h264 --disable-x86asm --enable-shared --enable-libmfx --enable-encoder=h264_qsv --enable-decoder=h264_qsv --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib
  libavutil      56. 30.100 / 56. 30.100
  libavcodec     58. 53.101 / 58. 53.101
  libavformat    58. 28.102 / 58. 28.102
  libavdevice    58.  7.100 / 58.  7.100
  libavfilter     7. 56.101 /  7. 56.101
  libswscale      5.  4.101 /  5.  4.101
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[h264 @ 0x55b706e18980] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x55b706e18980] decode_slice_header error
[h264 @ 0x55b706e18980] no frame!
[... 63 lines cut ...]
[sdp @ 0x55b706e11c40] RTP: missed 2 packets
[h264 @ 0x55b707027540] error while decoding MB 49 43, bytestream -222462.8kbits/s speed=   1x    
[h264 @ 0x55b707027540] concealing 160 DC, 160 AC, 160 MV errors in I frame
[sdp @ 0x55b706e11c40] max delay reached. need to consume packetrate=2453.8kbits/s speed=   1x    
[sdp @ 0x55b706e11c40] RTP: missed 5 packets
[h264 @ 0x55b707027540] error while decoding MB 74 40, bytestream -24
[h264 @ 0x55b707027540] concealing 375 DC, 375 AC, 375 MV errors in I frame
[sdp @ 0x55b706e11c40] max delay reached. need to consume packetrate=2445.1kbits/s speed=   1x    
[sdp @ 0x55b706e11c40] RTP: missed 2 packets
[h264 @ 0x55b707132340] error while decoding MB 49 43, bytestream -142445.2kbits/s speed=   1x    
[h264 @ 0x55b707132340] concealing 160 DC, 160 AC, 160 MV errors in I frame
[sdp @ 0x55b706e11c40] max delay reached. need to consume packetrate=2451.8kbits/s speed=   1x    
[sdp @ 0x55b706e11c40] RTP: missed 4 packets
[h264 @ 0x55b707027540] error while decoding MB 42 41, bytestream -16
[h264 @ 0x55b707027540] concealing 327 DC, 327 AC, 327 MV errors in I frame
frame=401511 fps= 30 q=-0.0 size= 4013056kB time=03:43:03.60 bitrate=2456.4kbits/s speed=   1x    
Socket error Event: 32 Error: 10053.
Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(192.168.0.77) at 15:50:19.

Type `help' to learn how to use Xshell prompt.
[E:\~]$

wangyan-intel, 2020-05-18:
@ailumiyana  Could you please provide your clip for checking? Thanks.

ailumiyana, 2020-05-28:
> @ailumiyana Could you please provide your clip for checking? Thanks.

sorry, I lost the clip . but i fix this problem by porting my application to the windows environment.

the clip likes a bare h264  stream that gop size 30 and no b-frames.

XinfengZhang, 2020-07-08:
@kanade2010  could we close the issue now, seems it is a clip based issue. if you have any further concern , please reopen it , thanks

closed 2020-07-08 by XinfengZhang