intel/media-driver #760: CSC p010->nv12 failed under 4K resolution

sunxinpeng, 2019-10-23:
This issue has been reported on FFmpeg trac, the link of the ticket:
[https://trac.ffmpeg.org/ticket/7764](url)

The cmdline to reproduce this issue in ffmpeg is:
```
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt p010 -s:v 3840x2160 -i input_4K_P010.yuv -vf 'hwupload,scale_vaapi=format=nv12,hwdownload,format=nv12' -f rawvideo -vsync passthrough  -y output.yuv
```
I tracked the code in media-driver and the key code that result in this failure is as follow:
1. 
```
bool VPHAL_VEBOX_STATE::VeboxIs2PassesCSCNeeded(
    PVPHAL_SURFACE              pSrc,
    PVPHAL_SURFACE              pRenderTarget)

if (IS_COLOR_SPACE_BT2020_YUV(pSrc->ColorSpace))
    {
        if ((pRenderTarget->ColorSpace == CSpace_BT601)           ||
            (pRenderTarget->ColorSpace == CSpace_BT709)           ||
            (pRenderTarget->ColorSpace == CSpace_BT601_FullRange) ||
            (pRenderTarget->ColorSpace == CSpace_BT709_FullRange) ||
            (pRenderTarget->ColorSpace == CSpace_stRGB)           ||
            (pRenderTarget->ColorSpace == CSpace_sRGB))
        {
            b2PassesCSCNeeded = true;
        }
    }
```
2. 
```
bool CompositeState::BuildFilter(
    PVPHAL_COMPOSITE_PARAMS         pCompParams,
    PKdll_FilterEntry               pFilter,
    int32_t*                        piFilterSize)

        //--------------------------------
        // Composition path does not support conversion from BT2020 RGB to BT2020 YUV, BT2020->BT601/BT709, BT601/BT709 -> BT2020
        //--------------------------------
        if (IS_COLOR_SPACE_BT2020_RGB(pSrc->ColorSpace)   &&
            IS_COLOR_SPACE_BT2020_YUV(pCompParams->Target[0].ColorSpace))  //BT2020 RGB->BT2020 YUV
        {
            eStatus = MOS_STATUS_UNIMPLEMENTED;
        }
        else if (IS_COLOR_SPACE_BT2020(pSrc->ColorSpace) &&
                 !IS_COLOR_SPACE_BT2020(pCompParams->Target[0].ColorSpace)) //BT2020->BT601/BT709
        {
            eStatus = MOS_STATUS_UNIMPLEMENTED;
        }
        else if (!IS_COLOR_SPACE_BT2020(pSrc->ColorSpace) &&
                 IS_COLOR_SPACE_BT2020(pCompParams->Target[0].ColorSpace))  //BT601/BT709 -> BT2020
        {
            eStatus = MOS_STATUS_UNIMPLEMENTED;
        }

        if (eStatus == MOS_STATUS_UNIMPLEMENTED)
        {
            goto finish;
        }
```
In this case, the pSrc->ColorSpace is BT2020 and the pRenderTarget->ColorSpace is BT709.

FurongZhang, 2019-11-20:
Please provide the driver log as we discussed, thanks in advance!

sunxinpeng, 2019-11-20:
> Please provide the driver log as we discussed, thanks in advance!

Attached the log file. Thanks.
[vp_dump.zip](https://github.com/intel/media-driver/files/3867629/vp_dump.zip)

sunxinpeng, 2019-12-05:
Kindly ping. Any progress?

FurongZhang, 2019-12-09:
@sunxinpeng I checked this issue. It is from P010 bt2020 to NV12 BT709 which render path can not support. Is it possible for you to set the scaling flag as VA_FILTER_SCALING_DEFAULT and have a try? If this flag, it will use Fixed Function to do the conversion.

sunxinpeng, 2019-12-09:
> @sunxinpeng I checked this issue. It is from P010 bt2020 to NV12 BT709 which render path can not support. Is it possible for you to set the scaling flag as VA_FILTER_SCALING_DEFAULT and have a try? If this flag, it will use Fixed Function to do the conversion.

I have tried this flag and the output will be black, which is the same as issue #763

uartie, 2020-02-24:
FWIW, this happens when the application sends VAProcColorStandardNone in the VP color standard settings.

But since:
```
commit 73bcc3415e32fa0697e384aa1f301a8b56327bc1
Author: Xinpeng Sun <xinpeng.sun@intel.com>
Date:   Mon Jan 13 15:33:28 2020 +0800

    Add VAProcColorStandardExplicit for input and output color standard query.
```
... ffmpeg will choose to use VAProcColorStandardExplicit, which fixes this issue.

And, as in #763, we also changed gstreamer to avoid VAProcColorStandardNone which fixed #763, too.

sunxinpeng, 2020-02-25:
> FWIW, this happens when the application sends VAProcColorStandardNone in the VP color standard settings.
>
> But since:
>
> ```
> commit 73bcc3415e32fa0697e384aa1f301a8b56327bc1
> Author: Xinpeng Sun [...]

IMHO, it seems to be a workaround fix for this issue in ffmpeg and gstreamer. It still need to be fixed in the driver, or maybe other applications will fail when they use iHD driver to do the same CSC for 4K resolution.

uartie, 2020-02-25:
> IMHO, it seems to be a workaround fix for this issue in ffmpeg and gstreamer. It still need to be fixed in the driver, or maybe other applications will fail when they use iHD driver to do the same CSC for 4K resolution.

Yes, I absolutely agree 😃

FurongZhang, 2020-04-02:
We have fixed this issue, please use the latest driver to verify it. a39fe9b.

FurongZhang, 2020-04-02:
I closed the issue, please let me know if any issues after verifying.

closed 2020-04-02 by FurongZhang