intel/media-driver #761: Excessive reallocation of buffers during HW-accelerated MPV playback

tokyovigilante, 2019-10-23:
I'm using mpv on Fedora 31 with Vulkan and VAAPI HW acceleration, and finding that libplacebo seems to be responsible for very excessive reallocation of frame buffers/slabs during playback. I initially experienced this as a significant memory leak due to bugs in the Sway compositor (https://github.com/swaywm/sway/issues/4662) however this has been fixed and the issue persists.

Verbose mpv logs show:
```
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.016 DS: 2.220/5 Dropped: 4
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.015 DS: 2.235/5 Dropped: 4
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.016 DS: 2.231/5 Dropped: 4
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.015 DS: 2.245/5 Dropped: 4
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.014 DS: 2.241/5 Dropped: 4
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Allocating 5160960 of imported memory from fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[vo/gpu/vulkan/libplacebo] Freed slab of size 5160960 from imported fd: 37
[statusline] AV: 00:00:02 / 03:47:00 (0%) A-V:  0.014 DS: 2.255/5 Dropped: 4
```
Repeated throughout playback. The buffer size is an even multiple of the video size/bit depth so I assume they are frame buffers. This issue only occurs with VAAPI-accelerated playback and is not seen with SW decoding. 

My system:
* X1 Yoga Gen 4 (i7-8665U w/ UHD 620 iGPU)
* Fedora 31 beta
* Sway/wlroots Wayland compositor (-git)
* Mesa 19.3-git (using new Iris driver and intel-media-driver although issue also seen with old i965 driver/libva driver).
* Kernel 5.4rc3
* mpv-git

mpv.conf:
```
fs=yes
hwdec=auto
vo=gpu
gpu-api=vulkan
video-sync=display-resample
profile=gpu-hq
interpolation
tscale=oversample
```

I've discussed this with the mpv Wayland developer (https://github.com/swaywm/sway/pull/4667#issuecomment-545521941) who suggested it is probably a driver issue but it does occur with both the i965 and new Iris drivers so I have also posted this over at the libplacebo tracker. Thanks!

LaserEyess, 2019-10-23:
I can reproduce this as well

My setup:
 * Intel i7-7500U CPU @ 2.70GHz w/ Intel HD Graphics 620
 * Arch Linux
 * sway version 1.2-43bd8dc4 
 * wlroots 0.8.1.r13.gd113e48a-1
 * mesa 19.2.1 (using iris driver)
 * intel-media-driver 19.3.0
 * Linux 5.3.7
 * mpv 0.29.0-883-g9dead2b932
 * libplacebo 1.18.0

The minimal reproducible command for mpv is:
```mpv --no-config --hwdec=vaapi --gpu-api=vulkan```

You get the same allocation messages for any video, always equal to the size of the window times bitdepth.

philipl, 2019-10-25:
As discussed in the libplacebo issue tracker, this is harmless. It's just logging too verbosely. These are the VA surface imports and not separate allocations

tokyovigilante, 2019-10-26:
Thanks for the comments @philipl, closing this one, the libplacebo fix has quieted the logs.

closed 2019-10-26 by tokyovigilante