intel/media-driver #763: [gst-vaapi][vp] P010 convert to other format will be black image for 4K resolution only.

feiwan1, 2019-10-24:
When use gst-vaapi to convert P010 to other format, the output YUV/RGB always black.

$ gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,width=3840,height=2160,format=P010_10LE ! vaapipostproc ! video/x-raw,format=BGRA ! filesink location=test.rgba

$ gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,width=3840,height=2160,format=P010_10LE ! vaapipostproc ! video/x-raw,format=NV12 ! filesink location=test.nv12

Check the output yuv file test.rgba/test.nv12 will show black picture. Expected output should show color bar image.

And when using other resolution less than 4K, the output looks good. For example 1080P:
$ gst-launch-1.0 videotestsrc num-buffers=1 ! video/x-raw,width=1920,height=1080,format=P010_10LE ! vaapipostproc ! video/x-raw,format=NV12 ! filesink location=test.nv12

This issue may have the same root cause with #760

FurongZhang, 2020-02-13:
With a merged change vaapipostproc: set input/output colorimetry in gst-vaapi, this issue can be fixed.

uartie, 2020-02-24:
It seems BGRA use-case is still an issue.

FurongZhang, 2020-04-02:
We have fixed this issue, please use the latest driver to verify it. a39fe9bc051a8c3efa8f35122a1585981ec7f816.

FurongZhang, 2020-04-02:
I closed the issue, please let me know if any issues after verifying.

closed 2020-04-02 by FurongZhang