intel/media-driver #766: [FFmpeg][QSV/VAAPI][HEVC][Encode] recommendation on default B frame numbers for better quality

fulinjie, 2019-10-30:
As mentioned in ffmpeg issue trac #8328:
https://trac.ffmpeg.org/ticket/8328#comment:10

The default "bf" value (b frame number) influences the encode quality.

> I tested different -bf values:
2: PSNR=41
4: PSNR=41
6: PSNR=41
7: PSNR=41
8: PSNR=25
12: PSNR=25
16: PSNR=25

For VAAPI hevc encoder, the default parameter for bf is 2:
​https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/vaapi_encode_h265.c#L1265
For QSV hevc encoder, the default parameter for bf is 8:
​https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/qsvenc_hevc.c#L261
According to x265 docs, it defaults to 4:
​https://x265.readthedocs.io/en/default/cli.html#slice-decision-options

> My small testing seems to indicate that current default value may be badly chosen for more demanding content.
Could you discuss this with the driver team, what is their recommendation on these nowadays?

So:
1.Is there any recommendations on the default b frame number in driver?
2.What about AVC or other codecs?

XinfengZhang, 2020-07-08:
for different usage scenario , should have different recommendation setting , bf is just one factor affecting quality. it should be a big topic.  if we just talking about bf, @Xiaogangli-intel  do you have any recommendation?

Xiaogangli-intel, 2021-09-22:
I will close it because there's no responses for more than 1 year.

closed 2021-09-22 by Xiaogangli-intel