intel/media-driver #768: [AVC VDEnc] Target usage changes with free media-driver

AntonGrishin, 2019-11-01:
Hi, 

original issue is https://github.com/Intel-Media-SDK/MediaSDK/issues/1735

short description 
W/ free/non-free driver we have different AVC low power encoding quality
```
./sample_encode h264   -w 176 -h 144 -f 30 -cqp -qpi 30 -qpp 30 -qpb 30 -qsv-ff    -i AUD_MW_E.yuv -o AUD_MW_E.264
``` 
Analysis from original issue:

> Issue reproduced with latest media-driver master.
> By driver dumps:
> With free driver sets TU = 7 by default, instead of TU = 4 w/ non-free.
> I found that with -DENABLE_NONFREE_KERNELS=OFF, also enables flag _FULL_OPEN_SOURCE.
> 
> This code is cause of issue:
> 
> ```
> #ifdef _FULL_OPEN_SOURCE
>     if (!GFX_IS_PRODUCT(m_encodeCtx->pMediaCtx->platform, IGFX_ICELAKE_LP))
>     {
>         if (m_encodeCtx->targetUsage >= 1 && m_encodeCtx->targetUsage <= 2)
>         {
>             m_encodeCtx->targetUsage = 4;
>         }
>         else if (m_encodeCtx->targetUsage >= 3 &&m_encodeCtx->targetUsage <= 5)
>         {
>             m_encodeCtx->targetUsage = 7;
>         }
>     }
> #endif
> ```
> I checked with m_encodeCtx->targetUsage = 4; in second else if block. Files w/ free/non-free drivers are bit match.

As I know, VDEnc not depends on kernels (we used CQP mode)
So why media-driver changes this field?
Also, based on condition in `if` - why TU doesn`t change only for ICL?

Regards,
Anton

XinfengZhang, 2019-11-05:
it is because ICL LP full open source driver need kernel for HME when it is target usage 1. but now , there are no such limitation----HME media kernel already was open sourced. will change it soon.

dvrogozh, 2019-11-05:
HME kernel was open sourced only for ICL. However, I still think that we should not remap target usages even for those platforms where HME is not open sourced (BDW/SKL): is that possible to just skip the kernel for TU1 if it is not available? I guess this should make quality(TU1)=quality(TU4)?

daleksan, 2020-08-20:
When you plan to merge #1013 ?

PR #1055 (2020-09-22): [AVC VDEnc] Fix target usage mapping issue for free media driver
Correct the mapping of target useage for free driver. Fixed #768.
commit 4c00c688 (2020-09-22): [AVC VDEnc] Fix target usage mapping issue for free media driver

Correct the mapping of target useage for free driver. Fixed #768.