intel/media-driver #771: Feature request: Expose VP9 encode support on Kabylake+ with the iHD driver

Brainiarc7, 2019-11-07:
Perhaps related to #630 and #205  : Will this ever be implemented?
A while back, FFmpeg received an [implementation for](http://git.ffmpeg.org/gitweb/ffmpeg.git/commit/33583803e107b6d532def0f9d949364b01b6ad5a) the [VP9 QSV encoder](http://git.ffmpeg.org/gitweb/ffmpeg.git/commit/fcbfdeeabe21cb0925313dab6079c50318a7bc71), with two constraints:

(a). Only low-power encoding is supported, and
(b). Icelake IGPs+ are supported. Older generations, even these capable of VP9 encoding with VAAPI are left out.

XinfengZhang, 2019-12-06:
@Brainiarc7  sorry for late response, we will not enable VP9 encode on Gen9 platforms , if someone still want it, he could try PR #717 , it add VP9 encode support. but Gen9 VME design is not optimized for VP9. so we have concerns on it.

Brainiarc7, 2019-12-06:
I see.
Perhaps a branch for the same could be created for legacy support.

XinfengZhang, 2019-12-06:
you mean to create a gen9_vp9_poc branch?  and sync the code regularly ?

Brainiarc7, 2019-12-06:
Yes.

Brainiarc7, 2019-12-11:
@XinfengZhang can that be done?
The patchwork is quite good and it works, as tested on CFL.

rosetta-jpn, 2020-01-29:
+1 to this feature.
Processors at a lot of existing devices are gen9.
i965 driver needs to be used on them due to a lack of iHD driver's VP9 encoding support.
i965 driver is no longer be maintained actively.
So it is nice to enable us to use iHD driver on gen9 devices.

sreerenjb, 2020-01-29:
@Brainiarc7  Thank you for testing this on CFL :)

anonimuos, 2020-09-16:
+1. VP9 should be exposed on Gen9 GPUs even with their limitations. On Windows also as there are many devices vs Ice Lake which is far too new.

fubsan, 2020-12-20:
Is there an update on this wrt roadmapping and 10th gen KBL derivative comet lake support?  Particularly considering this used to work with the intel vaapi driver on related cpu generations and also replies like this: https://github.com/intel/intel-vaapi-driver/issues/499

grzegorz-kurzejamski, 2021-01-29:
definitely +1 
At the project I'm working in we had to switch to 10th gen for multiple of processing nodes and it was quite a suprise that there is a 9/10th gen VP9 encoder support black hole with intel-vaapi-driver support up to 8th gen and iHD driver from 10th+ onwards. We were using VP9 Enc extensively up till now on 7th and 8th gen so this new generation is like a big downgrade, given better CPU specs and the very same iGPU. If there was a chance to build even minimal support for VP9 Encoder for 10th gen it would be much appreciated.

thetooth, 2021-02-15:
Also in the same situation with a 11th gen processor. Why advertise VP9 encoding support if nothing can access it?

XinfengZhang, 2021-02-17:
I could not commit that we could support VP9 VME(dual path) path on Gen9. there are several reasons, 1. VP9 VME path need the media-kernel (shader), and it was deprecated in future platforms , and it is not the direction , you could treat it as a temporal solution, TBH, I have no much confidence on it, so we have no plan to maintain it. 2. intel-vaapi-driver support all platforms of gen10- , so , if someone want to try VP9, we suggest they still keep older intel-vaapi-driver. 

#717 is a patch which support VP9 dual pipe path, it should be workable , maybe we could create a branch basing one release branch, but suppose we will not rebase it any more. from this perspective, seems it is similar with using intel-vaapi-driver.

XinfengZhang, 2021-02-17:
@thetooth , not sure whether I could answer your question, you mean you could not access ICL? , ICL 11th gen GPU certainly support VP9 encoding , it is fix function. hw support ..

karoldob, 2021-03-24:
Hi, 
This is curious that Intel says "VP9 supported for gen 10" and it is not true for Linux.
Gen11 is not available on market, gen8 is quite old. It is very sad :(
I hope that priority of this task will increase sooner than later...

karoldob, 2021-06-30:
@XinfengZhang It looks that on Intel gen 11, Ubuntu 21.04, iHD driver 21.1.1, VP9 encoder is still unsupported.
So "ICL 11th gen GPU certainly support VP9 encoding" on Linux with intel media-driver is not true.

sreerenjb, 2021-07-01:
The iHD driver should have the vp9 encode support on ICL. Make sure you have Gu&HuC firmware in place to play with vp9 hardware encode (in case if it is missing).

karoldob, 2021-07-16:
To sum up: encoding "works" but output video is broken.

Details:
- OS: Ubuntu 21.04, lastest kernel: 5.11.0-22-generic, 
- Gu&HuC enabled by adding "enable_guc=2" in kernel boot command and /etc/modprobe.d
- kernel boot command with i915.force_probe=4c8b, because intel gpu is not recognized in /dev/dri without that.
- HW: i5-11400
- linux kernel loads this binaries (sudo dmesg --ctime |grep -i [hg]uc) (shoudn't be icl_guc*?
[drm] GuC firmware i915/tgl_guc_49.0.1.bin version 49.0 submission:disabled
[drm] HuC firmware i915/tgl_huc_7.5.0.bin version 7.5 authenticated:yes

Results: 
- vainfo gives output with VAProfileVP9Profile0: VAEntrypointEncSliceLP (great)
- encoding sample clip (https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/1080/Big_Buck_Bunny_1080_10s_1MB.mp4) into VP9 gives broken video - output frame attached:
[image]

So formerly encoder "works", but drivers are still not ready for VP9 video encoding

Brainiarc7, 2021-11-19:
Hello there @karoldob ,

You'll need to add the the `vp9_superframe` and the `vp9_raw_reorder` bitstream filters to get correct output with these encoder wrappers.

Try the command below and report back:

```sh
ffmpeg -loglevel info -hide_banner -y -hwaccel vaapi -hwaccel_output_format vaapi -hwaccel_device /dev/dri/renderD128 -i Big_Buck_Bunny_1080_10s_2MB.mp4 -bsf:v "vp9_raw_reorder,vp9_superframe" -c:v vp9_vaapi out.webm
```

karoldob, 2022-04-26:
Hi @Brainiarc7 ,
I've tested few times and adding flags to bitstream filter not resolved the issue :( 
Last tests were on ubuntu 22.04, `ffmpeg` and `intel-media-va-driver-non-free` driver from default apt repo.
Result (image) looks the same (bad) as in my previous post

feiwan1, 2022-07-15:
This should be the issue when set rc_mode as ICQ, if specific rc_mode as CQP/CBR/VBR not this issue.