intel/media-driver #772: Encoding h264 stream with multiple slices reporting GPU hang

avstrakhov, 2019-11-11:
Encoding h264 stream with multiple slices reporting GPU hang

Software Stack:
libva 95eb8cf469367b532b391042fa0e77ca513ac94e refs/heads/master  https://github.com/01org/libva
gmmlib 94306f5443b97d9762dbb35ba65b341ee3e91cf0 refs/heads/master  https://github.com/intel/gmmlib
intel-media-driver 2e533df1562d9327d0847843bb71c31c7d2ea203 refs/heads/master  https://github.com/intel/media-driver
intel-mdeiasdk 3de3cbf0b05ee82965da8747aebf136858c9ddbb refs/heads/master https://github.com/Intel-Media-SDK/MediaSDK

Platform: SKL machine with 16 GB of memory running CentOS Linux 7 with kernel 4.14.20. 
Unmodified sample_fei application runs correctly when configured to use one slice, but fails to pak when 2 (or more) slices specified using command line:

./sample_fei -i ~/fish.yuv  -w 1980 -h 1080 -encpak -l 2 -o ~/tmp/fish.h264

Error message reported by intel-media-driver:

...
frame: 0  t:193 193 : submit [CODEC]:CRITICAL - GetStatusReport:3697: No reports available, wCurrIndex = 1, wFirstIndex = 1
ENC synced : 0
[LIBVA]:CRITICAL - StatusReport:264: Something unexpected happened in HW, return error to application

[ERROR], sts=MFX_ERR_GPU_HANG(-21), EncPakOneFrame, FEI PAK: SyncOperation failed at /home/astrahov/msdk/samples/sample_fei/src/fei_encpak.cpp:1044

[ERROR], sts=MFX_ERR_ABORTED(-12), Run, Unexpected error!! at /home/astrahov/msdk/samples/sample_fei/src/pipeline_fei.cpp:1755

[ERROR], sts=MFX_ERR_ABORTED(-12), RunPipeline, pPipeline->Run failed at /home/astrahov/msdk/samples/sample_fei/src/sample_fei.cpp:1336

Xiaogangli-intel, 2020-09-11:
@avstrakhov, is the issue still exist?

Xiaogangli-intel, 2021-09-22:
close it because there's no response for more than 1 year.

closed 2021-09-22 by Xiaogangli-intel