intel/media-driver #774: gen12: provide sources for the EU kernels or remove non-free kernels from the free build

dvrogozh, 2019-11-14:
https://github.com/intel/media-driver/blob/388f8b2144b88d8995bd43768f0d68b74a7e36d4/media_driver/agnostic/gen12/codec/media_srcs.cmake#L22

There are few driver build paths corresponding to -DENABLE_KERNELS=ON|OFF and -DENABLE_NONFREE_KERNELS=ON|OFF options. It seems that gen12 code upstreaming breaks free driver build squeezing non-free kernels inside. Could you, please, fix?

sebastinas, 2019-12-24:
The same probably also applies to to `kernelisa`.

dvrogozh, 2020-02-04:
It seems this issue was actually fixed by https://github.com/intel/media-driver/commit/9a872764e6f210e163579c82e4900ed8cd757b55 in master branch and by (cherry-picked)  https://github.com/intel/media-driver/commit/ea049d7da1cade3752ebd9e0a950d1264000bc18 in intel-media-19.4 branch. If so, then most recent release https://github.com/intel/media-driver/releases/tag/intel-media-19.4.0r should have this fix inside.

@XinfengZhang: can you, please, confirm that issue is fixed and close it?

@sebastinas, @tjaalton: FYI

XinfengZhang, 2020-02-11:
double comfirmed, it was fixed by 9a87276 , closed it

closed 2020-02-11 by XinfengZhang