intel/media-driver #776: [FFmpeg][VAAPI][HEVC][KBL/SKL][ 8K encoding] only 4K is reported to be supported in query surface attribute

fulinjie, 2019-11-15:
**Description:**
Reported Hardware does not support encoding at size 7680x4320 (constraints: width 32-4096 height 32-4096) on KBL/SKL.

**Log:**
[2019-11-12 06:48:07.239971] INFO: slash: CALL: ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v 7680x4320 -r:v 30 -i /opt/media/src/assets/otc-media/yuv/Peru_8K_HDR_7680x4320_100frames_I420.yuv -vf 'format=nv12,hwupload' -c:v hevc_vaapi -profile:v main -rc_mode CBR -g 30 -slices 3 -bf 0 -b:v 5000k -maxrate 5000k -low_power 0 -vframes 50 -y /opt/media/test-results/vaapi-fits/encode/hevc/07f6eb8a-0518-11ea-b408-0242ac110005_0/_0.test.ffmpeg-vaapi.encode.hevc/cbr/Peru_8K_HDR_7680x4320_100frames-cbr-main-30-30-3-0-5000k-5000k-0.h265 (pid: 17755)
[2019-11-12 06:48:07.242922] INFO: slash: ffmpeg version N-95710-gb25b6432a7 Copyright (c) 2000-2019 the FFmpeg developers
[2019-11-12 06:48:07.243031] INFO: slash:   built with gcc 7 (Ubuntu 7.4.0-1ubuntu1~18.04.1)
[2019-11-12 06:48:07.243316] INFO: slash:   configuration: --prefix=/opt/media/install --disable-static --enable-shared --enable-libdrm --enable-vaapi --enable-libmfx --disable-amf --disable-audiotoolbox --disable-cuda --disable-cuda-sdk --disable-cuvid --disable-d3d11va --disable-dxva2 --disable-libnpp --disable-mmal --disable-nvdec --disable-nvenc --disable-omx --disable-omx-rpi --disable-rkmpp --disable-v4l2-m2m --disable-vdpau --disable-videotoolbox --enable-gpl --enable-libx264 --enable-libx265
[2019-11-12 06:48:07.243385] INFO: slash:   libavutil      56. 35.101 / 56. 35.101
[2019-11-12 06:48:07.243446] INFO: slash:   libavcodec     58. 61.100 / 58. 61.100
[2019-11-12 06:48:07.243503] INFO: slash:   libavformat    58. 34.101 / 58. 34.101
[2019-11-12 06:48:07.243559] INFO: slash:   libavdevice    58.  9.100 / 58.  9.100
[2019-11-12 06:48:07.243613] INFO: slash:   libavfilter     7. 66.100 /  7. 66.100
[2019-11-12 06:48:07.243669] INFO: slash:   libswscale      5.  6.100 /  5.  6.100
[2019-11-12 06:48:07.243735] INFO: slash:   libswresample   3.  6.100 /  3.  6.100
[2019-11-12 06:48:07.243794] INFO: slash:   libpostproc    55.  6.100 / 55.  6.100
[2019-11-12 06:48:07.244558] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: VA-API version 1.6.0
[2019-11-12 06:48:07.244660] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: va_getDriverName() returns 0
[2019-11-12 06:48:07.244742] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: User requested driver 'iHD'
[2019-11-12 06:48:07.244835] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: Trying to open /opt/media/install/lib/dri/iHD_drv_video.so
[2019-11-12 06:48:07.245158] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: Found init function __vaDriverInit_1_6
[2019-11-12 06:48:07.246645] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] libva: va_openDriver() returns 0
[2019-11-12 06:48:07.246742] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] Initialised VAAPI connection: version 1.6
[2019-11-12 06:48:07.246831] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] VAAPI driver: Intel iHD driver - 1.0.0.
[2019-11-12 06:48:07.246927] INFO: slash: [AVHWDeviceContext @ 0x557b90c9e7c0] Driver not found in known nonstandard list, using standard behaviour.
[2019-11-12 06:48:07.262988] INFO: slash: [rawvideo @ 0x557b90cdd500] Estimating duration from bitrate, this may be inaccurate
[2019-11-12 06:48:07.263147] INFO: slash: Input #0, rawvideo, from '/opt/media/src/assets/otc-media/yuv/Peru_8K_HDR_7680x4320_100frames_I420.yuv':
[2019-11-12 06:48:07.263226] INFO: slash:   Duration: 00:00:03.33, start: 0.000000, bitrate: 11943937 kb/s
[2019-11-12 06:48:07.263333] INFO: slash:     Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p, 7680x4320, 11943936 kb/s, 30 tbr, 30 tbn, 30 tbc
[2019-11-12 06:48:07.265642] INFO: slash: Stream mapping:
[2019-11-12 06:48:07.265751] INFO: slash:   Stream #0:0 -> #0:0 (rawvideo (native) -> hevc (hevc_vaapi))
[2019-11-12 06:48:07.265815] INFO: slash: Press [q] to stop, [?] for help
[2019-11-12 06:48:07.266281] INFO: slash: [graph 0 input from stream 0:0 @ 0x557b90ceeb80] w:7680 h:4320 pixfmt:yuv420p tb:1/30 fr:30/1 sar:0/1 sws_param:flags=2
[2019-11-12 06:48:07.266380] INFO: slash: [auto_scaler_0 @ 0x557b90cf1bc0] w:iw h:ih flags:'bicubic' interl:0
[2019-11-12 06:48:07.266510] INFO: slash: [Parsed_format_0 @ 0x557b90ced400] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'Parsed_format_0'
[2019-11-12 06:48:07.267549] INFO: slash: [auto_scaler_0 @ 0x557b90cf1bc0] w:7680 h:4320 fmt:yuv420p sar:0/1 -> w:7680 h:4320 fmt:nv12 sar:0/1 flags:0x4
[2019-11-12 06:48:07.309419] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Input surface format is nv12.
[2019-11-12 06:48:07.309565] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Using VAAPI profile VAProfileHEVCMain (17).
[2019-11-12 06:48:07.309655] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[2019-11-12 06:48:07.309751] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Using VAAPI render target format YUV420 (0x1).
[2019-11-12 06:48:07.309823] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] RC mode: CBR.
[2019-11-12 06:48:07.309901] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] RC target: 100% of 5000000 bps over 1000 ms.
[2019-11-12 06:48:07.309979] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] RC buffer: 5000000 bits, initial fullness 3750000 bits.
[2019-11-12 06:48:07.310044] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] RC framerate: 30/1 (30.00 fps).
[2019-11-12 06:48:07.310122] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Using intra and P-frames (supported references: 3 / 1).
[2019-11-12 06:48:07.310201] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] Encoding pictures with 3 slices (default size 45 block rows).
[2019-11-12 06:48:07.310284] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] All wanted packed headers available (wanted 0xd, found 0x1f).
[2019-11-12 06:48:07.310381] INFO: slash: [hevc_vaapi @ 0x557b90cea2c0] **Hardware does not support encoding at size 7680x4320 (constraints: width 32-4096 height 32-4096).**
[2019-11-12 06:48:07.310496] INFO: slash: Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[2019-11-12 06:48:07.311386] INFO: slash: [AVIOContext @ 0x557b90ceab00] Statistics: 0 seeks, 0 writeouts
[2019-11-12 06:48:07.311480] INFO: slash: [AVIOContext @ 0x557b90ce5c40] Statistics: 49766400 bytes read, 0 seeks
[2019-11-12 06:48:07.311533] INFO: slash: Conversion failed!

**Analysis:**
https://github.com/intel/media-driver/blob/f030c7920b331ea83b5562ac914a3a09cda91342/media_driver/linux/common/ddi/media_libva_caps.cpp#L2661

And simply set the Max constraint to 8k would lead to **vaCreateContext faliure**:

Failed to create encode pipeline context: 19 (**resolution not supported**)

XinfengZhang, 2019-12-06:
@fulinjie just like we discussed offline, for HW and media kernel, 8K resolution could be handled , but it does not meet the RT requirement , I will change readme , thanks

fulinjie, 2019-12-06:
@XinfengZhang Got it, thanks.

XinfengZhang, 2019-12-06:
#795

fulinjie, 2019-12-25:
Fixed through https://github.com/intel/media-driver/pull/795, closed

closed 2019-12-25 by fulinjie