intel/media-driver #780: Intel media driver installation Error

itssivashankar, 2019-11-19:
As now i am working on ffmpeg transcoder process.we are trying to use intel quick sync with ffmpeg for hardware acceleration and to optimize cpu usage.
      I followed this documentation 
            https://github.com/intel/media-driver/tree/master
      As of now i had problem to init iHD driver in ubuntu 19.04.so kindly guide me to fix this problem.
[image]

dmitryermilov, 2019-11-19:
It refers to incorrect driver name (and possibly path). 
Please do:
export LIBVA_DRIVER_NAME="iHD"
export LIBVA_DRIVERS_PATH="/path/to/directory/with/iHD_drv_video.so"

dmitryermilov, 2019-12-06:
@itssivashankar , did it help?

closed 2020-07-08 by XinfengZhang