intel/media-driver #784: Valgrind annotations and invalid memory accesses

eero-t, 2019-11-20:
I'm seeing Valgrind annotations in the media-driver code:
```
$ git grep -l VALGRIND
media_driver/linux/common/os/i915/mos_bufmgr.c
media_driver/linux/common/os/i915/xf86drmMode.c
media_driver/linux/ult/libdrm_mock/mos_bufmgr_mock.c
media_driver/linux/ult/libdrm_mock/xf86drmMode_mock.c
```

But there are still a lot of Valgrind warnings for media-driver about invalid memory access when running test case from https://github.com/intel/media-driver/issues/757.

Either media-driver is buggy, annotations are missing from relevant places, or Valgrind annotations aren't automatically enabled although build environment includes Valgrind headers.  There's no CMake option for enabling them either.

XinfengZhang, 2019-12-06:
@eero-t thanks for reporting, we have notify the issue when we debug another issue reported by you 
#715 , we are cleaning memory violation, will have some commit soon. thanks again

XinfengZhang, 2020-07-08:
@eero-t , most of the violation is noise, it happens with GPU memory access. valgrind could not know the real gpu size

closed 2020-07-08 by XinfengZhang

eero-t, 2020-07-13:
Just tell Valgrind what the allocation sizes are, like e.g. libdrm does:
https://github.com/freedesktop/mesa-drm/blob/master/intel/intel_bufmgr_gem.c#L1485

Please re-open until this is properly fixed (I don't seem to be able to do that).