intel/media-driver #790: Add DRI3 Support

Master0ne, 2019-11-22:
> @akarshanbiswas said:
> For now you need to ask intel vaapi driver devs to add support for dri3. They had a PR but it is closed due to inactivity unfortunately.

I am trying to get HW video acceleration to work with [(ungoogled-)Chrom(ium) in GNOME on Wayland](https://github.com/jstkdng/ungoogled-chromium-archlinux/issues/6), which is what the (ungoogled-)chromium-ozone build is destined for, but it has been mentioned by someone with more insight that it may have to do with missing DRI3 support by the Intel driver.

I apologize for this unprofessional comment, I'm not a developer or programmer, but just a user.

CC: @jstkdng

akarshanbiswas, 2019-11-22:
Refer: https://github.com/intel/intel-vaapi-driver/pull/369

Apparently this PR was closed due to inactivity and I'm not sure if iHD driver has dri3 support or not to work through xWayland.

XinfengZhang, 2019-12-06:
media driver will not add support for display, so , no wayland support now. suppose DRI3 support should be added in vaapi level , such as https://github.com/intel/libva/pull/180?

akarshanbiswas, 2019-12-07:
> media driver will not add support for display, so , no wayland support now. suppose DRI3 support should be added in vaapi level , such as [intel/libva#180](https://github.com/intel/libva/pull/180)?

Interesting. I wonder why that PR is closed as well. :/

uartie, 2019-12-07:
> Interesting. I wonder why that PR is closed as well. :/

Those patches (intel/libva#180 and intel/intel-vaapi-driver#369) just need someone to take new ownership (i.e resubmit and follow through).  Volunteers?

akarshanbiswas, 2019-12-08:
@uartie If no one takes it, I can try if you want(rebase and resubmit). However I don't know how everything works in libva so I will follow blindly.

Also there is an existing problem of libva unable to initialize the driver when run through xwayland if LIBVA_DRIVER_NAME is not set. I hope https://github.com/intel/libva/pull/180 will solve that as well.

Jexu, 2022-12-02:
@XinfengZhang does it support from libva https://github.com/intel/libva/pull/614?

XinfengZhang, 2023-01-28:
yes, we could close the PR , thanks 
new libva 2.17.0 support DRI3.

closed 2023-01-28 by XinfengZhang