intel/media-driver #794: Non low-power HEVC encoding results in GPU hang when MaxSliceSize is set on ICL

xhaihao, 2019-12-05:
gmmlib: ce77e6ecb3b383b67b2b2624306219165fd1cfc3
media-driver: 11e07d6bf54a31472eb2624795478a01593bac37
MediaSDK: 195f95e9c5bfa890e8b8f8c7d593b99ff7839d89

command line:
/usr/share/mfx/samples/sample_encode h265 -mss 10000 -i input.yuv -o output.hevc -w 1920 -h 1080

Error message:
[ERROR], sts=MFX_ERR_GPU_HANG(-21), SynchronizeFirstTask, SyncOperation failed at /home/xhh/graphics/ex/gstreamer/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:153

[ERROR], sts=MFX_ERR_GPU_HANG(-21), GetFreeTask, m_TaskPool.SynchronizeFirstTask failed at /home/xhh/graphics/ex/gstreamer/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:2008

[ERROR], sts=MFX_ERR_GPU_HANG(-21), Run, m_pmfxENC->EncodeFrameAsync failed at /home/xhh/graphics/ex/gstreamer/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:2234

[ERROR], sts=MFX_ERR_GPU_HANG(-21), main, pPipeline->Run failed at /home/xhh/graphics/ex/gstreamer/MediaSDK/samples/sample_encode/src/sample_encode.cpp:1638

XinfengZhang, 2019-12-06:
@xhaihao ,  is it a MSDK issue or media_driver issue? has you checked whether VAEncMiscParameterMaxSliceSize is conveyed into driver?

xhaihao, 2019-12-09:
I think it is a driver issue, at least GPU hang should be avoided.

XinfengZhang, 2019-12-09:
need two attention:
1. VME + PAK mode does not support this feature, so there are no VA_ENC_SLICE_STRUCTURE_MAX_SLICE_SIZE is reported. @onabiull  please help to check whether MSDK already check this condition
2. media_driver should check the input , if VA_ENC_SLICE_STRUCTURE_MAX_SLICE_SIZE is not set. ParseMiscParamMaxSliceSize should return false instead of success

dmitryermilov, 2019-12-09:
> 1. VME + PAK mode does not support this feature, so there are no VA_ENC_SLICE_STRUCTURE_MAX_SLICE_SIZE is reported. @onabiull  please help to check whether MSDK already check this condition

Thank you. It's MSDK gap. Slice structure is hardcoded: https://github.com/Intel-Media-SDK/MediaSDK/blob/444ecb480a3918cc6faeb7662ca7492fd564f301/_studio/mfx_lib/encode_hw/h265/src/mfx_h265_encode_vaapi.cpp#L980 which shall be fixed

closed 2020-07-08 by XinfengZhang