intel/media-driver #796: 8k resolution support in encoders

alariono, 2019-12-06:
8k resolution support in encoders

If you looked on 
https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-encoding-input-format-and-max-resolution
HEVC 8bit Output NV12 Max Res. is 8k

BUT in https://github.com/intel/media-driver/blob/cdc808429e6aa5ece91935d20eccba2397db5a62/media_driver/linux/common/ddi/media_libva_caps.cpp#L1970 
max resolution still 4k...

        default:
            if (width > m_encMax4kWidth
                    || width < m_encMinWidth
                    || height > m_encMax4kHeight
                    || height < m_encMinHeight)
            {
                return VA_STATUS_ERROR_RESOLUTION_NOT_SUPPORTED;
            }
break;

Does this expected?

XinfengZhang, 2019-12-13:
it is an issue, 1. some platform just support 4K, but other support 8K, so need distinguish  it.

XinfengZhang, 2020-07-08:
issue should already been fixed, please refer to https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L813

closed 2020-07-08 by XinfengZhang