intel/media-driver #801: i965 vs iHD driver Stadia.

BramFr, 2019-12-11:
Hello,

I`m trying to use my debian installation to play Stadia. But everything i do i`ll get the vpx driver within chromium. 
Already installed chromium with vaapi driver. When i play a youtube video with VP9 codec everythings seems to be ok. When i open chrome://media-internals i can see i have mojo driver (gpu rendering). 

But when i start to play stadia it will fallback to VpxVideoDecoder

i965 (best playable with stadia)
```
vainfo
libva info: VA-API version 1.6.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.6 (libva 2.5.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

Here some chromium logs:
```
[9581:9581:1211/073209.808412:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[9584:9593:1211/073210.067761:ERROR:nss_util.cc(283)] After loading Root Certs, loaded==false: NSS error code: -8018
[9581:9581:1211/073210.077687:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
[9694:9694:1211/073212.906407:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[9694:9694:1211/073212.964476:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[9694:9694:1211/073212.973095:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[9694:9780:1211/073213.335125:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
[9694:9780:1211/073213.336442:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
[9694:9780:1211/073220.395329:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
[9694:9780:1211/073220.398739:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
[9694:9779:1211/073221.030374:ERROR:data_channel.cc(53)] Accepting maxRetransmits = -1 for backwards compatibility
[9694:9779:1211/073221.030526:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[9694:9779:1211/073226.441426:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[9694:9780:1211/073226.458719:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
[9694:9780:1211/073226.459871:ERROR:internal_decoder_factory.cc(56)] Trying to create decoder for unsupported format
```
Its looks like a unsupported format

After install iHD driver here some vainfo:
```
libva info: VA-API version 1.5.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri//iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_5
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.5 (libva 2.5.0)
vainfo: Driver version: Intel iHD driver - 1.0.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

Some chromium logs:
```
[8125:8125:1211/072732.324814:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264Main
[8125:8125:1211/072732.324872:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264High
[8125:8125:1211/072732.324885:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264ConstrainedBaseline
[8125:8125:1211/072732.365601:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[8125:8125:1211/072732.736689:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
[8129:8137:1211/072732.756579:ERROR:nss_util.cc(283)] After loading Root Certs, loaded==false: NSS error code: -8018
[8241:8241:1211/072736.879240:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[8241:8241:1211/072736.910585:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[8241:8241:1211/072736.914247:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[8241:8295:1211/072742.162519:ERROR:data_channel.cc(53)] Accepting maxRetransmits = -1 for backwards compatibility
[8241:8295:1211/072742.162646:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[8241:8295:1211/072742.779306:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[8125:8125:1211/072742.949565:ERROR:vaapi_wrapper.cc(1742)] Failed putting surface to pixmap VA error: unknown libva error
[8241:8246:1211/072742.969118:ERROR:h264_decoder_impl.cc(271)] avcodec_send_packet error: -1094995529
[8241:8295:1211/072748.724837:ERROR:data_channel.cc(53)] Accepting maxRetransmits = -1 for backwards compatibility
[8241:8295:1211/072748.724965:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[8241:8295:1211/072753.889222:ERROR:data_channel.cc(58)] Accepting maxRetransmitTime = -1 for backwards compatibility
[8241:8271:1211/072800.396415:ERROR:h264_decoder_impl.cc(271)] avcodec_send_packet error: -1094995529
[8547:8547:1211/072800.448401:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264Main
[8547:8547:1211/072800.448453:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264High
[8547:8547:1211/072800.448465:ERROR:vaapi_wrapper.cc(528)] GetConfigAttributes failed for va_profile VAProfileH264ConstrainedBaseline
[8547:8547:1211/072800.475141:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[8547:8547:1211/072800.511418:ERROR:buffer_manager.cc(488)] [.DisplayCompositor]GL ERROR :GL_INVALID_OPERATION : glBufferData: <- error from previous GL command
```

When i play with iHD the performance are really bad. When i`ll select the i965 driver i can play the game with some lags. But both still have VpxVideoDecoder with stadia.
Is there any way to get the mojo driver?

XinfengZhang, 2019-12-13:
thanks for reporting , just noticed the issue, both driver cant decoder the VP9 stream from stadia?
is there some special feature. ? you just said the bad perf, is there details?

BramFr, 2019-12-13:
Okey this is strange. 

i965:
This is stadia in the background and intel-gpu-top. 
[image]

iHD:
Stadia with intel-gpu-top.
[image]

Both streams give the VPX driver: 
When opening chromium and visit the page: chrome://media-internals 
[image]

BramFr, 2019-12-19:
After playing and trying the best settings. 
I cannot enable FPS on cloud gaming but i can give you a indication:

Game: Destiny 2
Best quality setting in Stadia
Running around in a aera without details.
- I965: 60 FPS
- iHD: 40 FPS
Running around in a aera with a lot of details.
- I965: 45 FPS (good to play but sometimes a lag)
- iHD: 10 FPS (not playable alof ot lag)

Balance quality/performance setting in Stadia
Running around in a aera without details.
- I965: 60 FPS
- iHD: 60 FPS
Running around in a area with a lot of details.
- I965: 60 FPS (Best play experience)
- iHD: 30 FPS (not playable alof ot lag)

kjozsa, 2020-01-21:
@BramFr did you find any new insights on this issue? 

I seem to have the same results on my Arch linux, VP9 with Youtube works with the MojoDecoder but Stadia uses the VpxVideoDecoder for some reason. 

My vainfo looks like this: 
`$ vainfo
vainfo: VA-API version: 1.6 (libva 2.6.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.4.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSlice
      VAProfileVP9Profile2            :	VAEntrypointVLD
`

sreerenjb, 2020-03-10:
There is no hardware acceleration in in linux-chromium :)
Unfortunately va-api based acceleration is only supported in ChromeOS.

BramFr, 2020-03-10:
@kjozsa did you install this? https://aur.archlinux.org/packages/chromium-vaapi/
Nope still have the VpxVideoDecoder but it works for playing a game.

kjozsa, 2020-03-10:
@BramFr I was running `chromium-vaapi-bin`, another AUR package on Arch, which was discontinued since then (and stopped working also).

XinfengZhang, 2020-07-08:
as @sreerenjb  mentioned https://github.com/intel/media-driver/issues/801#issuecomment-596842528, you are using sw decoder, not HW acceleration, right?

XinfengZhang, 2021-01-29:
please help to enable LIBVA_TRACE to ensure HW decoder works,  perf drop is not expected. the Video engine just ~10% percent.  so , there are no enough task is sent to media_driver and HW.  in another words, the bottleneck should not related with video decoder.

eero-t, 2023-07-04:
Maybe this can be closed as Stadia was shut down half a year ago: https://en.wikipedia.org/wiki/Google_Stadia ?

Unless there's some other use-case using `VpxVideoDecoder` and having same issue?

(in which case issue title should be updated to reflect that...)

Jexu, 2023-07-06:
If having no use-case, i will close this issue now. Please feel free to re-open it again if you all have any concerns.

closed 2023-07-06 by Jexu