intel/media-driver #805: [FFmpeg][AVC][VDENC][CBR/VBR][multi-slice]encode failed for CBR/VBR+VDENC+multislice

fulinjie, 2019-12-20:
Works in CQP, but fails in CBR/VBR mode.
CQP is fixed in:
https://github.com/intel/media-driver/pull/785

CMD:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v 720x480 -r:v 25 -i /root/media/testing/assets/yuv/720x480p.yuv -vf 'format=nv12,hwupload' -c:v h264_vaapi -profile:v high -rc_mode CBR -g 30 -slices 3 -b:v 500k -maxrate 500k -low_power 1 -vframes 150 -y out.h264

Log:

ffmpeg version N-96094-gaad0e26 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.3.0-16ubuntu3)
  configuration: --enable-vaapi --prefix=/opt/X11R7/iHD/ffmpeg_mainline --enable-nonfree --enable-shared --enable-libmfx --enable-gpl --enable-libx264 --enable-libx265 --disable-optimizations --enable-debug=3 --disable-stripping --enable-libvorbis --enable-libvpx --samples=fate-suite
  libavutil      56. 36.101 / 56. 36.101
  libavcodec     58. 65.100 / 58. 65.100
  libavformat    58. 35.101 / 58. 35.101
  libavdevice    58.  9.101 / 58.  9.101
  libavfilter     7. 69.101 /  7. 69.101
  libswscale      5.  6.100 /  5.  6.100
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
[AVHWDeviceContext @ 0x5614d879f740] libva: VA-API version 1.6.0
[AVHWDeviceContext @ 0x5614d879f740] libva: va_getDriverName() returns 0
[AVHWDeviceContext @ 0x5614d879f740] libva: User requested driver 'iHD'
[AVHWDeviceContext @ 0x5614d879f740] libva: Trying to open /opt/X11R7/iHD/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x5614d879f740] libva: Found init function __vaDriverInit_1_6
[AVHWDeviceContext @ 0x5614d879f740] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5614d879f740] Initialised VAAPI connection: version 1.6
[AVHWDeviceContext @ 0x5614d879f740] VAAPI driver: Intel iHD driver - 1.0.0.
[AVHWDeviceContext @ 0x5614d879f740] Driver not found in known nonstandard list, using standard behaviour.
[rawvideo @ 0x5614d87bec80] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from '/root/media/testing/assets/yuv/720x480p.yuv':
  Duration: 00:00:51.52, start: 0.000000, bitrate: 103680 kb/s
    Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p, 720x480, 103680 kb/s, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x5614d87ff5c0] w:720 h:480 pixfmt:yuv420p tb:1/25 fr:25/1 sar:0/1 sws_param:flags=2
[auto_scaler_0 @ 0x5614d8802a00] w:iw h:ih flags:'bicubic' interl:0
[Parsed_format_0 @ 0x5614d87fdb40] auto-inserting filter 'auto_scaler_0' between the filter 'graph 0 input from stream 0:0' and the filter 'Parsed_format_0'
[auto_scaler_0 @ 0x5614d8802a00] w:720 h:480 fmt:yuv420p sar:0/1 -> w:720 h:480 fmt:nv12 sar:0/1 flags:0x4
[h264_vaapi @ 0x5614d87f0b00] Input surface format is nv12.
[h264_vaapi @ 0x5614d87f0b00] Using VAAPI profile VAProfileH264High (7).
[h264_vaapi @ 0x5614d87f0b00] Using VAAPI entrypoint VAEntrypointEncSliceLP (8).
[h264_vaapi @ 0x5614d87f0b00] Using VAAPI render target format YUV420 (0x1).
[h264_vaapi @ 0x5614d87f0b00] RC mode: CBR.
[h264_vaapi @ 0x5614d87f0b00] RC target: 100% of 500000 bps over 1000 ms.
[h264_vaapi @ 0x5614d87f0b00] RC buffer: 500000 bits, initial fullness 375000 bits.
[h264_vaapi @ 0x5614d87f0b00] RC framerate: 25/1 (25.00 fps).
[h264_vaapi @ 0x5614d87f0b00] Using intra and P-frames (supported references: 3 / 0).
[h264_vaapi @ 0x5614d87f0b00] Slice count rounded up to 30 (from 3) due to driver constraints on slice structure.
[h264_vaapi @ 0x5614d87f0b00] Encoding pictures with 30 slices (default size 1 block rows).
[h264_vaapi @ 0x5614d87f0b00] All wanted packed headers available (wanted 0xd, found 0x1f).
[h264_vaapi @ 0x5614d87f0b00] Using level 3.
Output #0, h264, to 'out.h264':
  Metadata:
    encoder         : Lavf58.35.101
    Stream #0:0: Video: h264 (h264_vaapi) (High), 1 reference frame, vaapi_vld, 720x480, q=-1--1, 500 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.65.100 h264_vaapi
[MOS]:  CRITICAL - Mos_AddCommand:118: Unable to add command (no space).
[MHW]:  CRITICAL - AddMiStoreDataImmCmd:331: MOS returned error, eStatus = 0x22
[CODEC]:CRITICAL - EndStatusReport:3049: MOS returned error, eStatus = 0x22
[CODEC]:CRITICAL - ExecuteSliceLevel:1113: MOS returned error, eStatus = 0x22
[CODEC]:CRITICAL - ExecuteEnc:4443: Slice level encoding failed.
[CODEC]:CRITICAL - Execute:587: MOS returned error, eStatus = 0x22
[LIBVA]:CRITICAL - EncodeInCodecHal:1148: DDI:Failed in Codechal!
[LIBVA]:CRITICAL - EndPicture:81: DDI:DdiEncode_EncodeInCodecHal return failure.
[h264_vaapi @ 0x5614d87f0b00] Failed to end picture encode issue: 24 (internal encoding error).
[h264_vaapi @ 0x5614d87f0b00] Encode failed: -5.
Video encoding failed
[AVIOContext @ 0x5614d87fb5c0] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x5614d87ec5c0] Statistics: 518400 bytes read, 0 seeks
Conversion failed!

dmitryermilov, 2020-03-31:
We're going to revert #837 by (#895) because this change is wrong. HW/UMD in fact doesn't support VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS

dmitryermilov, 2020-03-31:
@fulinjie , can you please explain why ffmpeg wants VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS? What is the logic behind this?

fulinjie, 2020-03-31:
@dmitryermilov There is a query inside ffmpeg vaapi while initializing slice structure.
Media-driver returns the caps of VA_ENC_SLICE_STRUCTURE_EQUAL_ROWS hence ffmpeg vaapi would generate slices as much as the number of rows the bitstream has, regardless how much slices the user sets (for example -slices 3).

What's more, it leads to unexpected errors while encoding "equal row" slices.
 
https://github.com/intel-media-ci/ffmpeg/blob/master/libavcodec/vaapi_encode.c#L1898

XinfengZhang, 2020-07-08:
after https://github.com/intel/libva/pull/398 merged, ffmpeg should check this definition instead of "equal row"

fulinjie, 2020-07-08:
@XinfengZhang , yes "Equal rows" works for FFmpeg.
Feel free to close this issue even if it didn't got merged yet for now.

PR #837 (2020-02-19): media_libva_caps: add VAConfigAttribEncSliceStructure caps for VAEntrypointEncSliceLP

Add VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS for VAEntrypointEncSliceLP.

Fix #805.

Not sure whether it's proper to fix like this, please help to comment.
commit 11fd7a1d (2020-02-19): media_libva_caps: add VAConfigAttribEncSliceStructure caps for VAEntrypointEncSliceLP

Add VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS for VAEntrypointEncSliceLP.

Fix #805.
closed 2020-07-17 by XinfengZhang