intel/media-driver #808: 19.4.0 fails to builds with libva 2.5.0

sebastinas, 2019-12-24:
```
/<<PKGBUILDDIR>>/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp: In member function ‘VAStatus DdiEncodeHevc::ParseSeqParams(void*)’:           
/<<PKGBUILDDIR>>/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp:474:84: error: ‘struct _VAEncSequenceParameterBufferHEVC::<unnamed union>::<unna
med>’ has no member named ‘low_delay_seq’
  474 |     hevcSeqParams->LowDelayMode                       = seqParams->seq_fields.bits.low_delay_seq;
      |                                                                                    ^~~~~~~~~~~~~
/<<PKGBUILDDIR>>/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp:475:84: error: ‘struct _VAEncSequenceParameterBufferHEVC::<unnamed union>::<unna
med>’ has no member named ‘hierachical_flag’
  475 |     hevcSeqParams->HierarchicalFlag                   = seqParams->seq_fields.bits.hierachical_flag;
      |                                                                                    ^~~~~~~~~~~~~~~~
/<<PKGBUILDDIR>>/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp: In member function ‘virtual VAStatus DdiEncodeHevc::ParsePicParams(DDI_MEDIA_CO
NTEXT*, void*)’:                                                        
/<<PKGBUILDDIR>>/media_driver/linux/common/codec/ddi/media_ddi_encode_hevc.cpp:561:52: error: ‘VAEncPictureParameterBufferHEVC’ {aka ‘struct _VAEncPictureParam
eterBufferHEVC’} has no member named ‘hierarchical_level_plus1’
  561 |     hevcPicParams->HierarchLevelPlus1 = picParams->hierarchical_level_plus1;
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:7164: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_enc
ode_hevc.cpp.o] Error 1
```

Please either compile the code if only a sufficiently recent libva is available or bump the check in cmake.

EwanRoycroft, 2020-03-05:
Same issue on Ubuntu 19.10. Upgrading from libva 1.5.0 to 1.7.0 fixed it.

EwanRoycroft, 2020-03-05:
Same issue on Ubuntu 19.10. I made the change suggested by sebastinas and cmake then complained that I only had libva 1.5.0, despite having the libva2 (2.5.0) package installed.
After building libva from source, cmake said I had 1.7.0, and I was able to build without errors.

XinfengZhang, 2020-03-06:
1.7.0 is VAAPI version , libva version should be 2.7.0. libva is implementation of VAAPI

EwanRoycroft, 2020-03-06:
Okay, thanks for clearing that up.

commit 8a3f2758 (2020-03-24): Bump required libva version (fixes #808)
closed 2020-03-24 by intel-mediadev