intel/media-driver #813: Multi-GEN disable build failure

th0ma7, 2020-01-01:
While trying to compile a version specific to Gen9 I deactivated all other uneeded GENx such as
```
ifeq ($(findstring $(ARCH), 'apollolake'),$(ARCH))
CMAKE_ARGS += -DGEN8=OFF
CMAKE_ARGS += -DGEN9=ON
CMAKE_ARGS += -DGEN10=OFF
CMAKE_ARGS += -DGEN11=OFF
CMAKE_ARGS += -DGEN12=OFF
endif
```

Ended-up getting the following build error:
```
/home/th0ma7/git-ffmpeg-QS/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/cm/cm_execution_adv.cpp:46:24: fatal error: cm_hal_g11.h: No such file or directory
 #include "cm_hal_g11.h"
                        ^
compilation terminated.
make[4]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:531: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_execution_adv.cpp.o] Error 1
make[4]: Leaving directory '/home/th0ma7/git-ffmpeg-QS/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-19.4.0r/build'
make[3]: *** [CMakeFiles/Makefile2:168: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make[3]: Leaving directory '/home/th0ma7/git-ffmpeg-QS/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-19.4.0r/build'
make[2]: *** [Makefile:152: all] Error 2
make[2]: Leaving directory '/home/th0ma7/git-ffmpeg-QS/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-19.4.0r/build'
make[1]: *** [Makefile:74: intel-media_custom_compile] Error 2
make[1]: Leaving directory '/home/th0ma7/git-ffmpeg-QS/spksrc/cross/intel-media-driver'
make: [../../mk/spksrc.cross-cc.mk:138: arch-apollolake] Error 2 (ignored)
```

th0ma7, 2020-04-27:
Having a build issue with gen12 related to issue https://github.com/intel/media-driver/issues/907
In order to circumvent the problem I tried disabling `gen12` but had forgotten that disabling gpu arch doesn't work (e.g. problem still present with 20.1.1):
```[  8%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_execution_adv.cpp.o
/home/spksrc/git-ffmpeg-next/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/cm/cm_execution_adv.cpp:47:24: fatal error: cm_hal_g12.h: No such file or directory
 #include "cm_hal_g12.h"
                        ^
compilation terminated.
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:878: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_execution_adv.cpp.o' failed
make[4]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_execution_adv.cpp.o] Error 1
make[4]: Leaving directory '/home/spksrc/git-ffmpeg-next/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
CMakeFiles/Makefile2:162: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[3]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make[3]: Leaving directory '/home/spksrc/git-ffmpeg-next/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:151: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/spksrc/git-ffmpeg-next/spksrc/cross/intel-media-driver/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:90: recipe for target 'intel-media_custom_compile' failed
make[1]: *** [intel-media_custom_compile] Error 2
make[1]: Leaving directory '/home/spksrc/git-ffmpeg-next/spksrc/cross/intel-media-driver'
../../mk/spksrc.cross-cc.mk:109: recipe for target 'arch-apollolake' failed
make: [arch-apollolake] Error 2 (ignored)
```

```

XinfengZhang, 2020-07-17:
build system is designed to platform and feature independent, these part is less maintenance , always use one binary to cover all platform and all features. I set "reserved" flag , expect some effort from community to improve this part .

th0ma7, 2020-07-19:
As the option was documented I thought this as a nice way to reduce binary size on embedded systems.

XinfengZhang, 2020-07-19:
agree with you, it is designed for it. but as you know, we haven't enabled the test case to cover the combination , during this period there are some commits break it.

XinfengZhang, 2022-06-17:
https://github.com/intel/media-driver/wiki/Per-platform-build

closed 2022-06-17 by XinfengZhang

th0ma7, 2022-11-19:
@XinfengZhang I just want to say thnx for the change.  This allows me to keep our SynoCommunity driver updated while now able to disable newer gens on older Synology DSM which otherwise fails to build Xe usin gcc-4.9 such as:
```
# Disable Ice Lake (GEN11) & DG1/DG2 (GEN12)
# on older platforms
ifeq ($(call version_lt, $(TCVERSION), 7),1)
CMAKE_ARGS += -DGEN11=OFF
CMAKE_ARGS += -DGEN12=OFF
endif
```

bkuhls, 2022-11-19:
> fails to build Xe usin gcc-4.9 such as:

ftr: for my use case i need to build gen12 with gcc-4.9, to do so use this patch https://github.com/intel/media-driver/pull/1545 and these options
' ' ' 
	-DDG2=OFF \
	-DPVC=OFF \
	-DXE_LPG=OFF \
	-DXEHP_SDV=OFF \
	-DXe_M_plus=OFF \
	-DMTL=OFF \
' ' '