intel/media-driver #815: regression issue

xinyudox, 2020-01-07:
Command  line:

gst-launch-1.0 -vf filesrc location=./hevc/422/Allegro_HEVC_Main422_MT41_Deblocking_00_1920x1080_r2260.bin ! h265parse ! msdkh265dec ! videoconvert dither=0 ! video/x-raw,format=Y42B ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./Allegro_HEVC_Main422_MT41_Deblocking_00_1920x1080_r2260_1920x1080_422H.yuv

media-driver patch:

commit 108c0c720bf6a19b4e2ef817413f5aefc96e5b43
Author: Yan Wang <yan.wang@linux.intel.com>
Date:   Fri Dec 13 13:32:16 2019 +0800

    Check cabac_bypass_alignment_enabled_flag.
    
    Current HW cannot support HEVC stream with
    cabac_bypass_alignment_enabled_flag == 1.
    
    Signed-off-by: Yan Wang <yan.wang@linux.intel.com>

If revert this patch, the command line can pass, otherwise fail.

wangyan-intel, 2020-01-07:
@xinyudox Could you provide your video clip for checking? Thanks.

wangyan-intel, 2020-01-07:
I checked this clip which hasn't cabac_bypass_alignment_enabled_flag in SPS. So this patch should not impact this stream.  I find the gst pipeline use msdkh265dec. In fact, MSDK set cabac_bypass_alignment_enabled_flag by cabac_init_present_flag. It is **wrong**.
https://github.com/Intel-Media-SDK/MediaSDK/blob/master/_studio/shared/umc/codec/h265_dec/src/umc_h265_va_packer.cpp#L334
it will cause cabac_bypass_alignment_enabled_flag == 1 and fire the error handing by my driver patch. (108c0c7)
I have fixed it in https://github.com/Intel-Media-SDK/MediaSDK/pull/1891. 
You could apply this PR or change this line manually to try again.
Thanks.

wangyan-intel, 2020-01-07:
I have tried sample_decode with https://github.com/Intel-Media-SDK/MediaSDK/pull/1891. It can finish decoding and no error.

wangyan-intel, 2020-03-11:
This regression comes from MSDK bug. It has been fixed.

closed 2020-03-11 by wangyan-intel