intel/media-driver #816: VAAPI stopped working after latest update - Arch Linux

mahsoommoosa42, 2020-01-07:
I believe the issue started after the intel-media-driver package update to 19.4.0 on the 5th of Jan because it was working after setting LIBVA_DRIVER_NAME=iHD. 

Currently, the output of vainfo is
`DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [13]
param: 4, val: 0
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 18 (invalid parameter),exit`

However, I am able to get the correct output by using 
`vainfo --display drm --device /dev/dri/render129`
`vainfo: VA-API version: 1.6 (libva 2.6.0)
vainfo: Driver version: Intel iHD driver - 19.4.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI`

Any ideas what's happening here? I had been using VAAPI for my applications prior to the 5th. Is there any way to point VAAPI towards the correct render device?

XinfengZhang, 2020-01-08:
sorry about it, you had been using VAAPI prior to the 5th, but seems there are two display adapters, so intel igfx is renderD129 instead of renderD128, ls /dev/dri there should be two nodes , right?

mahsoommoosa42, 2020-01-08:
Yes.

mahsoommoosa42, 2020-01-09:
Hello. It seems it works occasionally between reboots.

mahsoommoosa42, 2020-01-10:
Solved by adding i915 to mkinitpcio. Seems to have been an issue with KMS Loading.

closed 2020-01-10 by mahsoommoosa42

arpu, 2020-11-19:
hi @mahsoommoosa42  

can you describe how to do this? ( i am on fedora 33 )

arpu, 2020-11-19:
i think this should be reopen

mahsoommoosa42, 2020-11-25:
@arpu refer to Early KMS Start section https://wiki.archlinux.org/index.php/kernel_mode_setting . I added i915 to MODULES and it solved the issue.