intel/media-driver #818: Segmenatation fault seen when running sample_decode with Intel Media Driver Q4'19 Release

ajitkumarkp, 2020-01-10:
Config:
- CML-S Ubuntu 18.04, Kerenl5.4
- Intel Media Driver Q4'19 Release which includes:
iHD Media driver: intel-media-19.4.0r 
gmmlib: intel-gmmlib-19.4.1
Libva: libva 2.6.0 (commit 5f693d9e60, v2.6-branch)

Command Line:
./sample_decode h264 -i ~/Downloads/AUD_MW_E.264 -r -vaapi -f 20 -d
pretending that stream is 30fps one
pretending that aspect ratio is 1:1
libva info: VA-API version 1.6.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
Decoding Sample Version 8.4.27.

Input video	AVC 
Output format	NV12
Input:
  Resolution	176x144
  Crop X,Y,W,H	0,0,176,144
Output:
  Resolution	176x144
Frame rate	30.00
Memory type		vaapi
MediaSDK impl		hw
MediaSDK version	1.31

Decoding started
Segmentation fault (core dumped)fread_fps: 0.000, fwrite_fps: 0.000

Issue is not seen when I upgrade to Libva master branch. The difference seems to be commit 7eec6f8f.

This may just require updating Libva version (libva 2.6.0) mentioned in https://github.com/intel/media-driver/releases. 

Please let me know if my observation is right or am I missing something. Thx.

brmarkus, 2020-01-15:
What does the core-dump file say, where does the crash happen?
If running in performance mode (i.e. not using "-r") (or writing into a file) then there is no crash anymore? Then it seems to be related to rendering.

brmarkus, 2020-01-15:
Does the crash happen with other files, too (also h.264, or h.265, MJPG, VC1, VP8, VP9, AV1 etc)?

ajitkumarkp, 2020-01-17:
> Does the crash happen with other files, too (also h.264, or h.265, MJPG, VC1, VP8, VP9, AV1 etc)?

Issue seen even with MPEG2.

ajitkumarkp, 2020-01-17:
> What does the core-dump file say, where does the crash happen?
> If running in performance mode (i.e. not using "-r") (or writing into a file) then there is no crash anymore? Then it seems to be related to rendering.

Issue is not seen when writing into a file (i.e -o out.yuv).

ajitkumarkp, 2020-01-17:
gdb (bt full) dump at the point to SIGSEGV attached.
[bt_full.txt](https://github.com/intel/media-driver/files/4078396/bt_full.txt)

AntonGrishin, 2020-01-20:
Issue same as https://github.com/Intel-Media-SDK/MediaSDK/issues/1916 
it`s regression in LibVA since intel/libva#351, which fixed in intel/libva#356

As I see on #351 - createDrawable is null, so when we try to call it, we got segfault.

ajitkumarkp, 2020-01-23:
@AntonGrishin Thanks for confirming. My initial observation as mentioned in the issue description was that the issue is not seen on libva master. However the main problem (although trivial) is that the  libva version for "Intel Media Driver Q4'19 Release" on https://github.com/intel/media-driver/releases still shows libva 2.6.0, and this needs to be fixed because many customers who do not follow CI need a stable BKC prescribed by Intel. If they use "Intel Media Driver Q4'19 Release" they will run into this issue.

XinfengZhang, 2020-03-08:
@ajitkumarkp thanks, I change it to 2.6.1

fredvb, 2020-07-12:
Issues seems to exist with  intel-media-driver 20.1.1-1 , intel-gmmlib 20.2.2-1 , libva 2.8 , libva-utils 2.8

I'm encountering the same issue when running MPV with --vo=vaapi when using iHD driver.
If I use i965 all goes well. 

My problem is that i965 doesn't work on a Comet Lake

Jexu, 2022-12-02:
Could you try latest libva and media driver, this should be fixed by https://github.com/intel/libva/pull/537

Jexu, 2023-03-03:
This issue should be fixed in latest libva and media driver, so I will close it now. Please re-open it again if you still meet this or having any other questions.

closed 2023-03-03 by Jexu